wolfSSL/wolfssh #89: How can I build this project for arch arm?

pangpangshui, 2018-08-09:
I built wolfssl but it gave me this error log:

```./testsuite/testsuite.test: line 117: /tes    tsuite/.libs/lt-testsuite.test: cannot execute binary file: Exec format erro  ```

JacobBarthelmeh, 2018-08-09:
Hi pangpangshui,

For cross compiling do not use sudo, this sets the ownership for the executables and anything built to be sudo user. I would also recommend building a static library and executables if cross compiling. Using "--enable-static --disable-shared" will remove the script wrappers for examples/tests created when using autotools, though that is not required.

The last thing is to make sure that the executable is being ran on the system it was compiled for.

Regards,
Jacob

JacobBarthelmeh, 2018-08-27:
Hi pangpangshui,

I'm going to go ahead and close this since it's been a couple weeks. You can reach out to us by email (www.wolfssl.com) or resurrect the case here if it's not resolved.

Regards,
Jacob

closed 2018-08-27 by JacobBarthelmeh