wolfSSL/wolfssh #95: wolfSSH error: Couldn't connect SSH stream

Vaibhav1587, 2018-08-27:
I have change the code a bit to connect to ssh with my server settings. But I got stuck at the following error.  
```
2018-08-27 16:46:09 [DEBUG] connect error: CLIENT_KEXDH_INIT_SENT, -17
wolfSSH error: Couldn't connect SSH stream
```

After searching and debugging I found that inside 

> src/internal.c

under `

> DoKexDhGexGroup

`
method ret is set to

>  WS_UNIMPLEMENTED_E

i.e. error code -17. 

can anybody tell me how to implement this feature ? or how to enable it?

ejohnstown, 2018-08-27:
At the beginning of DoKexDhGexGroup(), the variable `ret` is initialized to _WS_UNIMPLEMENTED_E_. Could you try changing that to WS_SUCCESS?

Vaibhav1587, 2018-08-28:
Changed it to WS_SUCCESS but screen halts.
Screen log attached.

[image]

ejohnstown, 2018-08-28:
Are you running the wolfSSH example client? Can you show me which command line options you are using for the client and echoserver?

Vaibhav1587, 2018-08-28:
Yes running he wolfSSH example client. 
Commands ,

Client : 

`./examples/client/client -h 192.168.xx.xxx -p 22 -u xxxxxx -P xxxxxxxxx`

Echoserver :

`./examples/echoserver/echoserver`

After long halt I get to know the error is Couldn't connect SSH stream. I have checked the port. not an issue with the port. 

[image]

ejohnstown, 2018-08-28:
wolfSSH's example echoserver listens on the non-priviledged port 22222. The echoserver and client are not using your system's username or authentication. The example usernames and passwords are found in the README file.

Vaibhav1587, 2018-08-29:
What if I don't use echoserver ? N run only client with twitch. 
I have taken the examples/client.c as a reference and set the host and port also username and password  according to my requirement. Now I'm expecting it to connect to my host but I'm failed to do it. It's selecting the algo's perfectly the initial key exchange is also perfect. I have implemented the same in JAVA. But I want it in C for STM32. 
Sceenshot of the log with error.. 

[image]

ejohnstown, 2018-08-30:
The example client is a trivial client example. It connects to a server and sends the string "Hello, wolfSSH!" to the peer. When using our echoserver as the SSH server, you should see "Server said: Hello, wolfSSH!"

This said, the error codes you are seeing at the tail of your log are telling me that the socket closed or failed somehow due to the -5, WS_CBIO_ERR_CONN_CLOSE. That's the I/O error code returned by the receive I/O callback function. It gets translated to -9, WS_SOCKET_ERROR, at the layer above the callback level.

TranslateReturnCode is returning a 0. In the UNIX-flavored environments, "If no messages are available to be received and the peer has performed an orderly shutdown, the value 0 is returned." We treat is as a closed socket.

ejohnstown, 2018-09-13:
We have fixed a couple of bugs in the KEX process in the last week or two. Could you try it again?

ejohnstown, 2019-02-21:
Are you still having an issue with the KEX process?

ejohnstown, 2019-03-11:
I am going to close this issue at this point.

closed 2019-03-11 by ejohnstown