wolfSSL/wolfssh #97: SSH implementation for STM32F103C8T6 Arm STM32

Vaibhav1587, 2018-09-03:
I need to implement the ssh for the above mention device. 
Any documentation or examples for that ?

ejohnstown, 2018-09-04:
Are you using the GCC arm cross-compiler and the configure script in the package? How are you setting up and building wolfSSL/wolfCrypt?

Vaibhav1587, 2018-09-06:
I have started up with the wolfCrypt test building with the reference to video posted on youtube link provided [https://www.youtube.com/watch?v=FG6jvINvls0](url) 
But I got stuck as its configuring the STM32F4 device and I'm using STM32F1

ejohnstown, 2018-09-13:
Which toolchain or IDE are you using? The YouTube video you refer to uses the Atollic TrueSTUDIO IDE, which is Eclipse based.

To build the libwolfssl.a library archive for the STM32F103C8, I opened the IDE/TRUESTUDIO/wolfssl directory as the project. I right-clicked on the the "wolfssl" in the Project Explorer pane and selected "Properties". In the project properties, I opened the "C/C++ Build:Settings" panel, and then selected the "Target Settings" tab. In the search field on the Target Settings I entered "STM32F103" and STM32F103C8 was listed under STM32F1:MCUs. I selected it and clicked "OK" at the bottom.

Next I added the following options to the file user_settings.h:
```
#define WOLFSSL_PUBLIC_MP
#define WOLFSSL_SHA512
#define WOLFSSL_SHA384
#define HAVE_AESGCM
#define WOLFCRYPT_ONLY
```

Last I built the project by first right-clicking on "wolfssl" in the Project Explorer pane and selected "Build Project". It finished successfully and in the debug folder in the Project Explorer was the file "libwolfssl.a". That would be used in wolfSSH.

Vaibhav1587, 2018-09-14:
Thanks!! That really helped. 

Its there any easiest way to run the echoserver on STM32F103C8T6 device and connect it from outside ? 
That would be simplest implementation of it.

ejohnstown, 2018-09-14:
The example echoserver can be embedded in another application. If you set the define flag "NO_MAIN_DRIVER" it doesn't build the main() function in echoserver.c. Then you can include echo server.c in another program and call echoserver_test(). You will need to make up your own argv and argc variables to pass to the function.

Vaibhav1587, 2018-09-21:
Nope. :( 

```
..\src\wolfssh\examples\echoserver\echoserver.c:25:33: fatal error: wolfssl/options.h: No such file or directory
     #include <wolfssl/options.h>
```

even when I have added the

>  libwolfssl.a

 file.

ejohnstown, 2018-09-21:
Did you add the wolfSSL header directory in your IDE's header search path? You select "Build Settings..." from the Project menu. Select "C/C++ General:Paths and Symbols" from the list box on the left of the properties window. On the "Includes" tab, you'll need to add the path to the wolfSSL header files.

ejohnstown, 2019-02-21:
Were you able to get the project to build?

closed 2019-03-04 by Vaibhav1587