wolfSSL/wolfssh #108: wolfSSH client on STM32F103C8T6

Vaibhav1587, 2018-09-21:
Run only the SSH client on the STM32F103C8T6 so that it can connect to my linux box.

ejohnstown, 2018-09-21:
Can I get one of our sales managers in touch with you to discuss setting up a quick start project? You'd get a few days of full time help getting wolfSSH running on your target.

Vaibhav1587, 2018-09-24:
We are already in talk with **Rod Weaver**. I'm actually checking the feasibility of application.

wuhanstudio, 2019-01-11:
Hi, I'm a master student from China, and I'm really interested with this idea. It'll be really convenient and cool if I can connect to servers in our lab from STM32F103C8T6. But actually I'm not sure whether or not it's possible with only 64KB flash. 

Here's my idea of one possible implementation. 
1. Network connection: STM32F103C8T6 + enc28j60 + lwip
2. RTOS with Terminal: [RT-Thread](https://github.com/rt-Thread/rt-thread/)
3. SSH Client: wolfCrypt + wolfSSH

Is it possible to set up a SSH client on STM32F103C8T6 in this way?
Thanks a lot for your time.

ejohnstown, 2019-02-21:
@Vaibhav1587: Were you able to (1) get it building or (2) do we have a support contract with you yet?
@wuhanstudio: I don't see why it wouldn't work. It would certainly be a tight fit in memory.

Vaibhav1587, 2019-02-27:
You can close this.

ejohnstown, 2019-02-27:
OK.

closed 2019-02-27 by ejohnstown