wolfSSL/wolfssh #109: `wolfSSH_stream_read` return value ignored in src/wolfsftp.c

nbaraz, 2018-09-29:
[Example](https://github.com/wolfSSL/wolfssh/blob/master/src/wolfsftp.c#L597) - This looks like a bug, since less data can be returned (or none at all in non-blocking mode), and then whatever was present in the buffer is interpreted as a number. This happens in multiple places.

Other functions (such as `SFTP_GetHeader` treat it as an error if less data is returned, which seems wrong since `wolfSSH_stream_read` doesn't guarantee to read exactly `bufSz` bytes.

I can open a pull request to fix this if this is indeed a bug :)

JacobBarthelmeh, 2018-10-08:
Hi nbaraz,

Thank you for the report. Not checking the return value is a bug. I added a fix for that in this PR (https://github.com/wolfSSL/wolfssh/pull/110).

For use with non blocking sockets the code will need some refactoring. Currently non blocking SFTP support is not yet supported, but if you want to submit code to add in support we could review it after a contribution agreement is signed. I'll add having non blocking support for SFTP to the desired feature request list.

Regards,
Jacob

JacobBarthelmeh, 2019-02-28:
Hi @nbaraz , the SFTP code got refactored for non blocking. Am going to go ahead and close this issue.

closed 2019-02-28 by JacobBarthelmeh