wolfSSL/wolfssh #124: Compiling wolfssh on RaspberryPi 3 Model B

yoctog, 2018-12-10:
Greetings,
The attached screen shot shows the error I received upon compiling wolfssh on raspberry pi 3. I have installed wolfssl with no problems. Your advices would be highly appreciated.

[image]

JacobBarthelmeh, 2018-12-11:
Thanks for the report yoctog! I was able to reproduce the case on a raspberrypi 2 we have in the office here. Will investigate into the warning.

JacobBarthelmeh, 2018-12-12:
yoctog, I opened a PR to change the types in our examples for fixing the warning. Could you tell us some about your use case with wolfSSH?

yoctog, 2018-12-12:
Thank you for your reply.
We are currently building detectors which should detect some run time modifications and we are testing the performance of detectors on workloads which uses wolfSSH. 
removing "Werror", same error is generated for server_test and echo_server_test

running `make check` --> testsuite.test (fail)
after running `make`, tried to run `echoserver` did not run I get command options list as output.
same thing with client and server examples.

closed 2018-12-12 by yoctog