wolfSSL/wolfssh #133: How can I build project in Windows?

Vaibhav1587, 2019-02-19:
Environment - 

> Windows 10 

> Visual Studio 2017

> wolfssl-3.15.3

> wolfssh-1.3.0

I have successfully build the wolfssl no issues. But while generating the build for wolfssh it fails 

```
D:\Project\keil_workspace\wolfssl\wolfssh\wolfssh.vcxproj : error  : The project file could not be loaded. Could not find file 'D:\Project\keil_workspace\wolfssl\wolfssh\wolfssh.vcxproj'.  D:\Project\keil_workspace\wolfssl\wolfssh\wolfssh.vcxproj

D:\Project\keil_workspace\wolfssl\api-test\api-test.vcxproj : error  : The project file could not be loaded. Could not find a part of the path 'D:\Project\keil_workspace\wolfssl\api-test\api-test.vcxproj'.  D:\Project\keil_workspace\wolfssl\api-test\api-test.vcxproj

D:\Project\keil_workspace\wolfssl\echoserver\echoserver.vcxproj : error  : The project file could not be loaded. Could not find a part of the path 'D:\Project\keil_workspace\wolfssl\echoserver\echoserver.vcxproj'.  D:\Project\keil_workspace\wolfssl\echoserver\echoserver.vcxproj

D:\Project\keil_workspace\wolfssl\unit-test\unit-test.vcxproj : error  : The project file could not be loaded. Could not find a part of the path 'D:\Project\keil_workspace\wolfssl\unit-test\unit-test.vcxproj'.  D:\Project\keil_workspace\wolfssl\unit-test\unit-test.vcxproj

D:\Project\keil_workspace\wolfssl\client\client.vcxproj : error  : The project file could not be loaded. Could not find a part of the path 'D:\Project\keil_workspace\wolfssl\client\client.vcxproj'.  D:\Project\keil_workspace\wolfssl\client\client.vcxproj

D:\Project\keil_workspace\wolfssl\wolfsftp-client\wolfsftp-client.vcxproj : error  : The project file could not be loaded. Could not find a part of the path 'D:\Project\keil_workspace\wolfssl\wolfsftp-client\wolfsftp-client.vcxproj'.  D:\Project\keil_workspace\wolfssl\wolfsftp-client\wolfsftp-client.vcxproj

```

Followed the instructions of having projects a same location without version or anything. Opened the wofssh.sln from wolfssh folder and trying to build.

kaleb-himes, 2019-02-19:
@Vaibhav1587,

Prior to building the wolfssl64.sln did you make sure to copy the contents of wolfssl/ide/winvs/user_setting.h to replace the wolfssl/IDE/WIN/user_settings.h?

If not the library and wolfSSH will be configured differently and could explain the errors you are seeing. Please try performing that step and let me know if that resolves the issues you are seeing.

- K

Vaibhav1587, 2019-02-19:
Thanks for quick reply. 

Done. It Worked 👍 

I have question about implementing the ssh on STM32 Nucleo F4463RE. Should I create new thread for that ?

kaleb-himes, 2019-02-20:
@Vaibhav1587,

You can either open a new thread or if you prefer for best response times and tracking purposes you can always email us at support@wolfssl.com or open a support inquiry directly through our zendesk portal at https://wolfssl.zendesk.com.

Happy to hear the recommendation resolved the issue you were seeing!

Cheers,

K

closed 2019-02-20 by kaleb-himes

Vaibhav1587, 2019-02-25:
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1181	cannot open input file 'wolfssl.lib'	wolfssh	D:\Project\keil_workspace\wolfssh\ide\winvs\wolfssh\LINK	1