wolfSSL/wolfssh #136: STM32F446RE wolfSSH setup

Vaibhav1587, 2019-02-27:
Environment : 
**Windows 10
Atollic TrueStudio
WolfSSH 1.3.0 **

I'm trying to build the application which can connect the linux box through ssh on said STM32 board. I have copied the source folders in my project and also the Free RTOS on the board. 

Following are the errors where I stuck,

```
In file included from ../Inc/../wolfssl/wolfcrypt/types.h:29:0,
                 from ../wolfssh/ssh.h:34,
                 from ..\Src\main.c:55:
../Inc/../wolfssl/wolfcrypt/settings.h:1769:14: warning: #warning "For timing resistance / side-channel attack prevention consider using harden options" [-Wcpp]
             #warning "For timing resistance / side-channel attack prevention consider using harden options"
              ^~~~~~~
In file included from ../Inc/../wolfssl/wolfcrypt/types.h:30:0,
                 from ../wolfssh/ssh.h:34,
                 from ..\Src\main.c:55:
../Inc/../wolfssl/wolfcrypt/wc_port.h:145:17: error: unknown type name 'pthread_mutex_t'
         typedef pthread_mutex_t wolfSSL_Mutex;
                 ^~~~~~~~~~~~~~~
In file included from c:\program files (x86)\atollic\truestudio for stm32 9.0.1\armtools\arm-atollic-eabi\include\dirent.h:7:0,
                 from ../Inc/../wolfssl/wolfcrypt/wc_port.h:339,
                 from ../Inc/../wolfssl/wolfcrypt/types.h:30,
                 from ../wolfssh/ssh.h:34,
                 from ..\Src\main.c:55:
c:\program files (x86)\atollic\truestudio for stm32 9.0.1\armtools\arm-atollic-eabi\include\sys\dirent.h:10:2: error: #error "<dirent.h> not supported"
 #error "<dirent.h> not supported"
  ^~~~~
In file included from ../Inc/../wolfssl/wolfcrypt/types.h:30:0,
                 from ../wolfssh/ssh.h:34,
                 from ..\Src\main.c:55:
../Inc/../wolfssl/wolfcrypt/wc_port.h:360:9: error: unknown type name 'DIR'
         DIR*   dir;
         ^~~
```

Vaibhav1587, 2019-02-27:
Got some success after changing the `setting.h in wolfssl/wolfcrypt/`

But now got issue,

> ../Inc/../wolfssl/wolfcrypt/settings.h:1133:44: fatal error: stm32f4xx_cryp.h: No such file or directory

both the stm32f4xx_cryp.h and stm32f4xx_hash.h are missing.

Vaibhav1587, 2019-02-27:
Resolved above issue also.

Now stuck on,

`../wolfssh/test.h:54:23: fatal error: netdb.h: No such file or directory`

ejohnstown, 2019-02-28:
That header file is one we include in the Linux/macOS/UNIX build. In test.h, we have 4 different sets of includes depending on 4 different build options, right now.

1. Windows
2. Microchip Harmony
3. Nucleus
4. Everything else

You are building for everything else, and don't have that file. That has the API for the function we use for IP address lookups.

Vaibhav1587, 2019-02-28:
Can I just copy `stm32f4xx_cryp.h and stm32f4xx_hash.h` from somewhere else ? does any other files also required ? I have created project using STM32CubeMX.

dgarske, 2019-02-28:
See issue in wolfSSL repo: 
https://github.com/wolfSSL/wolfssl/issues/2126

closed 2019-03-04 by Vaibhav1587