wolfSSL/wolfssh #137: WS_SELECT_RECV_READY undefined

Vaibhav1587, 2019-02-28:
Inside **`NonBlockSSH_connect(WOLFSSH* ssh)`**

Below values are undefined,
```
WS_SELECT_RECV_READY
WS_SELECT_ERROR_READY
WS_SELECT_TIMEOUT
```

Which files contains the definition?

ejohnstown, 2019-02-28:
Those constants are defined in wolfssh/test.h. They are used for the select() calls in the example tools.

Vaibhav1587, 2019-02-28:
I'm trying to write my own client.c 
For now skipped the above code and just running upto connect the server, 

getting following error,

```
3>main.c
3>d:\project\keil_workspace\wolfssl\examples\client\client.h(27): error C2061: syntax error: identifier 'client_test'
3>d:\project\keil_workspace\wolfssl\examples\client\client.h(27): error C2059: syntax error: ';'
3>d:\project\keil_workspace\wolfssl\examples\client\client.h(27): error C2059: syntax error: '<parameter-list>'
```

It showing error in the `**client.h** @ **THREAD_RETURN WOLFSSH_THREAD client_test(void* args);** `

ejohnstown, 2019-02-28:
In your file **main.c**, are you including the files:

    #include <wolfssh/ssh.h>
    #include <wolfssh/test.h>
    #include "examples/client/client.h"

Your compile might not like the `THREAD_RETURN` or `WOLFSSH_THREAD` labels. For the most part WOLFSSH_THREAD is nothing, but in the Windows case is `__stdcall`. `THREAD_RETURN` is OS specific.

ejohnstown, 2019-02-28:
Which RTOS are you using?

Vaibhav1587, 2019-02-28:
Now I'm trying to build it on WIN 10. 
 
Then will try for the device(FreeRTOS). Need to get it working first.

Vaibhav1587, 2019-02-28:
Win 10 with Visual Studio 2017 

My main : 

```
#include <stdio.h>

#define WOLFSSL_USER_SETTINGS
#define WOLFSSH_TEST_CLIENT
#include "wolfssh/ssh.h"
#include "wolfssh/test.h"
#include "D:/Project/keil_workspace/wolfssh/examples/client/client.h"
const char testString[] = "Hello, wolfSSH!";

static int SetEcho(int on)
{
#if !defined(USE_WINDOWS_API) && !defined(MICROCHIP_PIC32)
	static int echoInit = 0;
	static struct termios originalTerm;
	if (!echoInit) {
		if (tcgetattr(STDIN_FILENO, &originalTerm) != 0) {
			printf("Couldn't get the original terminal settings.\n");
			return -1;
		}
		echoInit = 1;
	}
	if (on) {
		if (tcsetattr(STDIN_FILENO, TCSANOW, &originalTerm) != 0) {
			printf("Couldn't restore the terminal settings.\n");
			return -1;
		}
	}
	else {
		struct termios newTerm;
		memcpy(&newTerm, &originalTerm, sizeof(struct termios));

		newTerm.c_lflag &= ~ECHO;
		newTerm.c_lflag |= (ICANON | ECHONL);

		if (tcsetattr(STDIN_FILENO, TCSANOW, &newTerm) != 0) {
			printf("Couldn't turn off echo.\n");
			return -1;
		}
	}
#else
	static int echoInit = 0;
	static DWORD originalTerm;
	HANDLE stdinHandle = GetStdHandle(STD_INPUT_HANDLE);
	if (!echoInit) {
		if (GetConsoleMode(stdinHandle, &originalTerm) == 0) {
			printf("Couldn't get the original terminal settings.\n");
			return -1;
		}
		echoInit = 1;
	}
	if (on) {
		if (SetConsoleMode(stdinHandle, originalTerm) == 0) {
			printf("Couldn't restore the terminal settings.\n");
			return -1;
		}
	}
	else {
		DWORD newTerm = originalTerm;

		newTerm &= ~ENABLE_ECHO_INPUT;

		if (SetConsoleMode(stdinHandle, newTerm) == 0) {
			printf("Couldn't turn off echo.\n");
			return -1;
		}
	}
#endif

	return 0;
}
byte userPassword[256];

static int wsUserAuth(byte authType,
	WS_UserAuthData* authData,
	void* ctx)
{
	const char* defaultPassword = (const char*)ctx;
	word32 passwordSz = 0;
	int ret = WOLFSSH_USERAUTH_SUCCESS;

	(void)authType;
	if (defaultPassword != NULL) {
		passwordSz = (word32)strlen(defaultPassword);
		memcpy(userPassword, defaultPassword, passwordSz);
	}
	else {
		printf("Password: ");
		SetEcho(0);
		if (fgets((char*)userPassword, sizeof(userPassword), stdin) == NULL) {
			printf("Getting password failed.\n");
			ret = WOLFSSH_USERAUTH_FAILURE;
		}
		else {
			char* c = strpbrk((char*)userPassword, "\r\n");;
			if (c != NULL)
				*c = '\0';
		}
		passwordSz = (word32)strlen((const char*)userPassword);
		SetEcho(1);
#ifdef USE_WINDOWS_API
		printf("\r\n");
#endif
	}

	if (ret == WOLFSSH_USERAUTH_SUCCESS) {
		authData->sf.password.password = userPassword;
		authData->sf.password.passwordSz = passwordSz;
	}

	return ret;
}

THREAD_RETURN WOLFSSH_THREAD client_test(void* args) {
	WOLFSSH_CTX* ctx = NULL;
	WOLFSSH* ssh = NULL;
	SOCKET_T sockFd = WOLFSSH_SOCKET_INVALID;
	SOCKADDR_IN_T clientAddr;
	socklen_t clientAddrSz = sizeof(clientAddr);
	char rxBuf[80];
	int ret;
	int ch;
	word16 port = wolfSshPort;
	char* host = (char*)wolfSshIp;
	const char* username = "jack";
	const char* password = "upthehill"                                                                                                                                                                                                                                                                                                                ;
	byte imExit = 0;
	byte nonBlock = 0;

	ctx = wolfSSH_CTX_new(WOLFSSH_ENDPOINT_CLIENT, NULL);
	if (ctx == NULL)
		err_sys("Couldn't create wolfSSH client context.");

	if (((func_args*)args)->user_auth == NULL)
		wolfSSH_SetUserAuth(ctx, wsUserAuth);
	else
		wolfSSH_SetUserAuth(ctx, ((func_args*)args)->user_auth);

	ssh = wolfSSH_new(ctx);
	if (ssh == NULL)
		err_sys("Couldn't create wolfSSH session.");

	if (password != NULL)
		wolfSSH_SetUserAuthCtx(ssh, (void*)password);

	ret = wolfSSH_SetUsername(ssh, username);
	if (ret != WS_SUCCESS)
		err_sys("Couldn't set the username.");

	build_addr(&clientAddr, host, port);
	tcp_socket(&sockFd);
	ret = connect(sockFd, (const struct sockaddr *)&clientAddr, clientAddrSz);
	if (ret != 0)
		err_sys("Couldn't connect to server.");

	printf("Connect to the server");
}

int main(int argc, char **argv) {
	func_args args;

	args.argc = argc;
	args.argv = argv;
	args.return_code = 0;
	args.user_auth = NULL;

	WSTARTTCP();
#ifdef DEBUG_WOLFSSH
	wolfSSH_Debugging_ON();
#endif

	wolfSSH_Init();

	ChangeToWolfSshRoot();
#ifndef NO_WOLFSSH_CLIENT
	client_test(&args);
#endif

	wolfSSH_Cleanup();

	return args.return_code;
}
```

Vaibhav1587, 2019-02-28:
Build log now, 

```
1>------ Rebuild All started: Project: wolfssl, Configuration: Debug Win32 ------
1>cl : Command line warning D9035: option 'Gm' has been deprecated and will be removed in a future release
1>wolfmath.c
1>wolfevent.c
1>wc_port.c
1>wc_encrypt.c
1>signature.c
1>sha512.c
1>sha256.c
1>sha.c
1>rsa.c
1>ripemd.c
1>random.c
1>rabbit.c
1>pwdbased.c
1>poly1305.c
1>pkcs7.c
1>pkcs12.c
1>memory.c
1>md5.c
1>md4.c
1>logging.c
1>Generating Code...
1>Compiling...
1>integer.c
1>hmac.c
1>hc128.c
1>hash.c
1>error.c
1>ecc.c
1>dsa.c
1>dh.c
1>des3.c
1>coding.c
1>chacha20_poly1305.c
1>chacha.c
1>camellia.c
1>blake2b.c
1>asn.c
1>arc4.c
1>aes.c
1>wolfio.c
1>tls13.c
1>tls.c
1>Generating Code...
1>d:\project\keil_workspace\wolfssl\wolfcrypt\src\asn.c(3186): warning C4701: potentially uninitialized local variable 'algId' used
1>d:\project\keil_workspace\wolfssl\wolfcrypt\src\asn.c(3192): warning C4701: potentially uninitialized local variable 'pkcs8KeySz' used
1>d:\project\keil_workspace\wolfssl\wolfcrypt\src\asn.c(3253): warning C4701: potentially uninitialized local variable 'pbeOidSz' used
1>Compiling...
1>ssl.c
1>ocsp.c
1>keys.c
1>internal.c
1>crl.c
1>Generating Code...
1>wolfssl.vcxproj -> C:\Users\vaibh\source\repos\SSH_Test\Debug\wolfssl.lib
1>Done building project "wolfssl.vcxproj".
2>------ Rebuild All started: Project: wolfssh, Configuration: Debug Win32 ------
2>internal.c
2>io.c
2>keygen.c
2>log.c
2>memory.c
2>port.c
2>ssh.c
2>Generating Code...
2>keygen.obj : warning LNK4221: This object file does not define any previously undefined public symbols, so it will not be used by any link operation that consumes this library
[... 100 lines cut ...]
3>wolfssl.lib(ecc.obj) : error LNK2019: unresolved external symbol _mp_invmod referenced in function _ecc_map
3>wolfssl.lib(rsa.obj) : error LNK2001: unresolved external symbol _mp_invmod
3>wolfssl.lib(ecc.obj) : error LNK2019: unresolved external symbol _mp_isodd referenced in function _ecc_projective_add_point
3>wolfssl.lib(ecc.obj) : error LNK2019: unresolved external symbol _mp_count_bits referenced in function _wc_ecc_sign_hash_ex
3>wolfssl.lib(rsa.obj) : error LNK2001: unresolved external symbol _mp_count_bits
3>wolfssl.lib(ecc.obj) : error LNK2019: unresolved external symbol _mp_rshb referenced in function _wc_ecc_sign_hash_ex
3>wolfssl.lib(ecc.obj) : error LNK2019: unresolved external symbol _mp_sqr referenced in function _ecc_map
3>wolfssl.lib(ecc.obj) : error LNK2019: unresolved external symbol _mp_montgomery_reduce referenced in function _ecc_map
3>wolfssl.lib(ecc.obj) : error LNK2019: unresolved external symbol _mp_montgomery_setup referenced in function _wc_ecc_mulmod_ex
3>wolfssl.lib(ecc.obj) : error LNK2019: unresolved external symbol _mp_div_2 referenced in function _ecc_projective_add_point
3>wolfssl.lib(ecc.obj) : error LNK2019: unresolved external symbol _mp_montgomery_calc_normalization referenced in function _wc_ecc_mulmod_ex
3>wolfssl.lib(asn.obj) : error LNK2019: unresolved external symbol _mp_leading_bit referenced in function _SetASNIntMP
3>wolfssl.lib(rsa.obj) : error LNK2019: unresolved external symbol _mp_2expt referenced in function _wc_CompareDiffPQ
3>wolfssl.lib(rsa.obj) : error LNK2019: unresolved external symbol _mp_to_unsigned_bin_len referenced in function _wc_RsaFunctionSync
3>wolfssl.lib(rsa.obj) : error LNK2019: unresolved external symbol _mp_set_int referenced in function _wc_CheckRsaKey
3>wolfssl.lib(rsa.obj) : error LNK2019: unresolved external symbol _mp_gcd referenced in function __CheckProbablePrime
3>wolfssl.lib(rsa.obj) : error LNK2019: unresolved external symbol _mp_lcm referenced in function _wc_MakeRsaKey
3>wolfssl.lib(rsa.obj) : error LNK2019: unresolved external symbol _mp_abs referenced in function _wc_CompareDiffPQ
3>wolfssl.lib(wolfmath.obj) : error LNK2019: unresolved external symbol _fp_zero referenced in function _mp_rand
3>wolfssl.lib(wolfmath.obj) : error LNK2019: unresolved external symbol _mp_lshd referenced in function _mp_rand
3>C:\Users\vaibh\source\repos\SSH_Test\Debug\SSH_Test.exe : fatal error LNK1120: 54 unresolved externals
3>Done building project "SSH_Test.vcxproj" -- FAILED.
========== Rebuild All: 2 succeeded, 1 failed, 0 skipped ==========

```

ejohnstown, 2019-02-28:
Which versions of wolfSSL and wolfSSH are you building with? The last released or fresh from GitHub?

The last released versions are:
* wolfSSL v3.15.7
* wolfSSH v1.3.0

Vaibhav1587, 2019-02-28:
Same using,

* wolfSSL v3.15.7
* wolfSSH v1.3.0

ejohnstown, 2019-02-28:
Pull the latest from GitHub. There has been some major updates to wolfSSH with regards to Windows builds and non-blocking. We will be doing a new release soon.

closed 2019-03-04 by Vaibhav1587