wolfSSL/wolfssh #138: Not able to find 'Dhkey'

Vaibhav1587, 2019-03-04:
Getting following issue,
`wolfssh/wolfssh/internal.h:245:9: error: unknown type name 'DhKey'`

whereas` #include <wolfssl/wolfcrypt/dh.h>` added.

my 

> user_settings.h

 is as follows,

```
#define NO_WOLFSSL_DIR
#define WOLFCRYPT_ONLY
#define WOLFSSL_KEY_GEN
#define HAVE_ECC
#define HAVE_AESGCM
#define HAVE_HASHDRBG
#define WOLFSSL_SHA384
#define WOLFSSL_SHA512
#define NO_PSK
#define NO_HC128
#define NO_RC4
#define NO_RABBIT
#define NO_DSA
#define NO_MD4
#define WC_RSA_BLINDING
#define USE_FAST_MATH
#define TFM_TIMING_RESISTANT
#define ECC_TIMING_RESISTANT
#define WOLFSSL_PUBLIC_MP 
```

ejohnstown, 2019-03-04:
Are you getting this error when every C file compiles or just a specific one? The DH option is one that is enabled by default and you would have to define something to disable it. You aren't getting an error about the compiler not finding the file wolfssl/wolfcrypt/dh.h either and it isn't giving you errors about the other structures. Is it?

Vaibhav1587, 2019-03-04:
Can you suggest me the MCU from STM32 in which I can deploy the code n it
will run without any issues.
I have following boards
STM32 Nucleo F446RE
STM32 Nucleo F303K8
STM32 Nucleo F767ZI
STM32 Nucleo L432KC
FRDM KL 82Z
FRDM K82F

I feel like none of the above is useful. Do I have to buy discovery board?
If yes then which ? Or any other board, MCU on which I can implement the
wolfssh without any issues.

dgarske, 2019-03-06:
Hi Vaibhav1587,

Do you have the pre-processor macro `WOLFSSL_USER_SETTINGS` defined?

As John said DH should be on and included by default assuming `NO_DH` is not defined.

Thanks,
David Garske, wolfSSL

Vaibhav1587, 2019-03-06:
Yes, I have defined it in settings.h

ejohnstown, 2019-03-06:
OK. I ran through the YouTube video you had linked before and it definitely left a lot of stuff out. It was produced earlier last year and some things have changed.

First, there are no special `#define` just to get STM32 builds to work. The STM32 options are used to take advantage of the on board crypto some of the chips have. If you don't set anything, then you get the algorithms from our code.

I will email you an archive with a readme file, a user_settings.h file, and an updated copy of tool's generated main.c file. It compiles to an output file for me here, but I don't have my STM32 prototype board or cables available to test it. The example should run the wolfCrypt test.

Normally for commercial customers, we sell consulting to help you get our code up and running in your environment in a couple days.

ejohnstown, 2019-03-11:
I am going to close this issue at this point.

closed 2019-03-11 by ejohnstown

Vaibhav1587, 2019-03-11:
Yeah sure.. I'm just looking into something else. But I will be return to
this project soon.

Thanks.