wolfSSL/wolfssh #141: echoserver: Error with SSH shutdown.

umokk, 2019-03-07:
_examples\echoserver_ and _tests\testsuite_ give "Error with SSH shutdown." when testing against _examples\client_.
Is it normal?

ejohnstown, 2019-03-07:
No, that is not normal. Can you tell me which commit you are using, the command lines used, and which OS? I'm using commit 8038d48 on macOS. I ran the testsuite without an error. When running the echo server with:

    $ ./examples/echoserver/echoserver

I can run both the wolfSSH client and OpenSSH's client against it without errors:

    $ ./examples/client/client -u jill
    $ ssh -p 22222 jill@localhost

--John

umokk, 2019-03-07:
Hi, John. Thank you for not ignoring my question. I'm using the latest commit of wolfssh with the latest commit of wolfssl.
First of all, sorry that I forgot to give info about my testing environment. My bad. I'm using Windows 10, so first I built wolfssl and wolfssh from the source and then ran the tests.
I tested wolfssh in the following way:
1. Run server.exe
2. Run client.exe in another cmd window. This test went as expected: after "Hello..." both programs exited.

1. Run echoserver.exe
2. Run client.exe in another cmd window. This time after "Hello..." client exited, echoserver continued to run, but spitted out "Error with SSH shutdown.". It says it every time a client connects to the server.

1. Run testsuite.exe
It spitted out 2 lines:
Hello...
Error with SSH shutdown.

That's totally understandable since testsuite uses echoserver.c.

ejohnstown, 2019-03-08:
It is just the timing between the client and the echoserver on the shutdown. The client sends the shutdown and then closes its socket. Meanwhile the server is trying to receive the shutdown and send its own, but it gets the FIN. I'll add this to the list of things to fix for the next release.

umokk, 2019-03-09:
Thank you. I guess it's a fix then.

closed 2019-03-09 by umokk