wolfSSL/wolfssh #142: Buffer mishandling based on receive callback value

DeanIsMe, 2019-03-14:
`static int Receive(WOLFSSH* ssh, byte* buf, word32 sz)` in internal.c calls the user receive callback `recvd = ssh->ctx->ioRecvCb(ssh, buf, sz, ssh->ioReadCtx);`. If this user callback was to return a negative value that's not handled in the `switch` statement, then  it will return that negative value `default:                return recvd;`

`static int GetInputData(WOLFSSH* ssh, word32 size)` only checks for 2 negative values: `-1` and `WS_WANT_READ`. Any other other negative values will be used in here as `in`:
`ssh->inputBuffer.length += in;
        inSz -= in;`
Which means we'll subtract from `inputBuffer.length` and increase `inSz` - which means the length could run negative which can lead to writing to areas outside of the buffer.

I confirmed this error by returning -100 in my receive callback function, which resulted in my program crashing (presumably due to writing to invalid memory location). Using bare metal PIC32 env.

JacobBarthelmeh, 2019-03-18:
Hi DeanlsMe, Thank you for the report! I'm investigating into this. For quicker response times on new reports there is the support channel at support @ wolfssl . com.

JacobBarthelmeh, 2019-03-18:
Treating this as a bug report and have a PR under review at https://github.com/wolfSSL/wolfssh/pull/144

closed 2019-03-28 by JacobBarthelmeh