wolfSSL/wolfssh #167: [SFTP] Handle size found was too big

bagder, 2019-05-13:
I have the beginnings of support for wolfSSH in curl in a [separate branch](https://github.com/curl/curl/tree/bagder/wolfssh).

I build and run my code on a Linux Debian unstable x64-64. I run an SFTP server with OpenSSH_7.9p1. Plain password authentication.

When I build with current wolfssh (commit a3881e1471383a486b527, dated May 10) and run my code like this:

`curl -u scptest:scptest sftp://localhost/~/hello -v`

It fails, and wolfssh logs the following lines from the SFTP open call and onwards, ending with it returning a failure that I just can't understand. Does this happen because of some mistake on my part or is this a problem in wolfssh?

Any suggestions on what I can do to analyze this further?

~~~
2019-05-13 15:42:48 [SFTP] Entering wolfSSH_SFTP_Open()
2019-05-13 15:42:48 [SFTP] SFTP OPEN STATE: INIT
2019-05-13 15:42:48 [SFTP] SFTP OPEN STATE: SEND
2019-05-13 15:42:48 [DEBUG] Entering wolfSSH_stream_send()
2019-05-13 15:42:48 [DEBUG] Entering SendChannelData()
2019-05-13 15:42:48 [DEBUG] Entering ChannelFind(): peer 0
2019-05-13 15:42:48 [DEBUG] Leaving ChannelFind(): 0x564930cd7500
2019-05-13 15:42:48 [DEBUG] BP: paddingSz = 14
2019-05-13 15:42:48 [DEBUG] EncryptAead aes128-gcm@openssh.com
2019-05-13 15:42:48 [INFO]   dataSz = 40
2019-05-13 15:42:48 [INFO]   peerWindowSz = 2097129
2019-05-13 15:42:48 [INFO]   update peerWindowSz = 2097089
2019-05-13 15:42:48 [DEBUG] Entering wolfSSH_SendPacket()
2019-05-13 15:42:48 [DEBUG] Embed Send trying to send 84
2019-05-13 15:42:48 [DEBUG] Embed Send sent 84
2019-05-13 15:42:48 [DEBUG] SB: Shrinking output buffer
2019-05-13 15:42:48 [DEBUG] Entering ShrinkBuffer()
2019-05-13 15:42:48 [DEBUG] SB: usedSz = 0, forcedFree = 0
2019-05-13 15:42:48 [DEBUG] SB: releasing dynamic buffer
2019-05-13 15:42:48 [DEBUG] Leaving ShrinkBuffer()
2019-05-13 15:42:48 [DEBUG] Leaving SendChannelData(), ret = 40
2019-05-13 15:42:48 [DEBUG] Leaving wolfSSH_stream_send(), txd = 40
2019-05-13 15:42:48 [SFTP] SFTP OPEN STATE: GETHANDLE
2019-05-13 15:42:48 [SFTP] Entering wolfSSH_SFTP_GetHandle
2019-05-13 15:42:48 [SFTP] SFTP GET HANDLE STATE: INIT
2019-05-13 15:42:48 [SFTP] SFTP GET HANDLE STATE: GET_HEADER
2019-05-13 15:42:48 [DEBUG] Entering wolfSSH_stream_read()
2019-05-13 15:42:48 [DEBUG] PR1: size = 16
2019-05-13 15:42:48 [DEBUG] Receive: recvd = 16
2019-05-13 15:42:48 [DEBUG] PR2: size = 52
2019-05-13 15:42:48 [DEBUG] Receive: recvd = 36
2019-05-13 15:42:48 [DEBUG] DecryptAead aes128-gcm@openssh.com
2019-05-13 15:42:48 [DEBUG] DoPacket sequence number: 12
2019-05-13 15:42:48 [DEBUG] Decoding MSGID_CHANNEL_DATA
2019-05-13 15:42:48 [DEBUG] Entering DoChannelData()
2019-05-13 15:42:48 [DEBUG] Entering ChannelFind(): self 0
2019-05-13 15:42:48 [DEBUG] Leaving ChannelFind(): 0x564930cd7500
2019-05-13 15:42:48 [DEBUG] Entering ChannelPutData()
2019-05-13 15:42:48 [INFO]   dataSz = 17
2019-05-13 15:42:48 [INFO]   windowSz = 16179
2019-05-13 15:42:48 [INFO]   windowSz = 16162
2019-05-13 15:42:48 [DEBUG] Leaving DoChannelData(), ret = -1057
2019-05-13 15:42:48 [DEBUG] PR3: peerMacSz = 16
2019-05-13 15:42:48 [DEBUG] PR4: Shrinking input buffer
2019-05-13 15:42:48 [DEBUG] Entering ShrinkBuffer()
2019-05-13 15:42:48 [DEBUG] SB: usedSz = 0, forcedFree = 1
2019-05-13 15:42:48 [DEBUG] SB: releasing dynamic buffer
2019-05-13 15:42:48 [DEBUG] Leaving ShrinkBuffer()
2019-05-13 15:42:48 [DEBUG] PR5: txCount = 336, rxCount = 1200
2019-05-13 15:42:48 [DEBUG] Leaving wolfSSH_stream_read(), rxd = 9
2019-05-13 15:42:48 [SFTP] SFTP GET HANDLE STATE: CHECK_REQ_ID
2019-05-13 15:42:48 [SFTP] SFTP GET HANDLE STATE: READ
2019-05-13 15:42:48 [DEBUG] Entering wolfSSH_stream_read()
2019-05-13 15:42:48 [DEBUG] Leaving wolfSSH_stream_read(), rxd = 8
2019-05-13 15:42:48 [SFTP] Handle size found was too big
2019-05-13 15:42:48 [SFTP] SFTP GET HANDLE STATE: CLEANUP
2019-05-13 15:42:48 [SFTP] Error getting handle
2019-05-13 15:42:48 [SFTP] SFTP OPEN STATE: CLEANUP
2019-05-13 15:42:48 [DEBUG] Entering wolfSSH_get_error()
* wolfssh SFTP open failed: -1004
* multi_done
* Connection #0 to host localhost left intact
curl: (79) wolfssh SFTP open failed: -1004
2019-05-13 15:42:48 [DEBUG] Entering wolfSSH_Cleanup()
2019-05-13 15:42:48 [DEBUG] Leaving wolfSSH_Cleanup(), returning 0

JacobBarthelmeh, 2019-06-24:
Hi bagder,

Thank you for the report. I'm looking into reproducing the problem. In the mean time if this is being induced by a call to wolfSSH_SFTP_Open please have the input argument "handleSz" set to a length of 256 with a matching "handle" input buffer that is at least 256 bytes (or WOLFSSH_MAX_HANDLE from wolfssh/wolfsftp.h). This is the max handle size that can be sent across according to the RFC and it may be that the buffer being passed in is too small to hold the handle found from the SFTP packet.

Regards,
Jacob

JacobBarthelmeh, 2019-07-05:
Hi @bagder,

After looking over the branch linked to I think setting the handleSz will resolve the issue.

In lib/wolfssh.c
```
507     case SSH_SFTP_DOWNLOAD_INIT:
(add)     sshc->handleSz = WOLFSSH_MAX_HANDLE; /* size of "handle" buffer in struct from lib/ssh.h */
508       rc = wolfSSH_SFTP_Open(sshc->ssh_session, sftp_scp->path,
509                              WOLFSSH_FXF_READ, NULL,
510                              sshc->handle, &sshc->handleSz);
```

~jacob

bagder, 2019-08-16:
Yes, that seems to fix it!

Can I suggest you document this somewhere? I was certainly not obvious to me...

closed 2019-08-16 by JacobBarthelmeh