wolfSSL/wolfssh #172: Possible init bug in BuildUserAuthRequestEcc

LinuxJedi, 2019-06-24:
rPad and sPad are set to 1 or 0 and then immediately set to 0 afterwards. Not sure which is the correct behaviour yet but I suspect this could be a bug.

JacobBarthelmeh, 2019-06-24:
Hi @LinuxJedi,

Is this in reference to the rPad and sPad in the wolfSSH file src/internal.c around line 6070? What version of wolfSSH is being used? 

Regards,
Jacob

LinuxJedi, 2019-06-24:
Hi,
Sorry, I should have narrowed it down more in my first message.

The ones around line 7080 of that file in master branch.

JacobBarthelmeh, 2019-06-24:
Got it, thanks LinuxJedi. I see the line of code now immediately after the ternary operation that sets both rPad and sPad to 0. Will investigate into it.

LinuxJedi, 2019-06-24:
That's the one. I would have submitted a PR but I've only had a glance at the code and am not sure yet what the correct behaviour is.

ejohnstown, 2019-06-24:
I did that one. I'll take a look.

JacobBarthelmeh, 2019-06-24:
@ejohnstown thanks for taking a look at it, there is a PR here with a possible change for review (https://github.com/wolfSSL/wolfssh/pull/174). Out of curiosity @LinuxJedi what tool was used to find these? Was it found with using scan-build?

LinuxJedi, 2019-06-24:
Yes, I used scan-build to find these. I hope it was helpful.

JacobBarthelmeh, 2019-06-24:
We appreciate the reports, it was helpful! And feel free to open any additional issues that you come across. Thank you for confirming the tool used, it will help us with knowing where we can improve continuous integration testing with wolfSSH.

closed 2019-06-24 by JacobBarthelmeh