wolfSSL/wolfssh #179: Possible copy/paste error in wolfsftp.c

LinuxJedi, 2019-07-03:
In wolfsftp.c function wolfSSH_SFTP_Remove() this repeated condition is used four times:

`if (ssh->error != WS_WANT_READ && ssh->error != WS_WANT_READ)`

I presume the condition should be WS_WANT_WRITE but didn't want to submit a patch until I was sure that was the intention.

(also found using cppcheck)

JacobBarthelmeh, 2019-07-05:
Hi LinuxJedi, That is correct it should have a check for WS_WANT_READ && WS_WANT_WRITE not the double check on WS_WANT_READ that it currently has.

LinuxJedi, 2019-07-05:
Cool, want me to patch it or have you got it?

JacobBarthelmeh, 2019-07-05:
I'll go ahead and add in a pull request for it and make reference to this issue number as the initial report. Thanks again LinuxJedi for the feedback!

LinuxJedi, 2019-07-05:
No problem. Glad I could help :)

closed 2019-08-05 by JacobBarthelmeh