wolfSSL/wolfssh #183: Putty client fails to connect to wolfssh example/server

csenanda, 2019-07-31:
Hello,
I am running example server from v1.4.0-stable-34-gf07b134 and using a putty client to connect.  The session was not successful and the client keeps expecting packets from server.

Below is the server log when it happens.

2019-07-31 22:23:41 [DEBUG] Leaving ShrinkBuffer()
2019-07-31 22:23:41 [DEBUG] PR5: txCount = 0, rxCount = 48
2019-07-31 22:23:41 [DEBUG] accept state: CLIENT_USERAUTH_REQUEST_DONE
2019-07-31 22:23:41 [DEBUG] BP: paddingSz = 10
2019-07-31 22:23:41 [DEBUG] CreateMac hmac-sha2-256
2019-07-31 22:23:41 [DEBUG] Encrypt aes128-cbc
2019-07-31 22:23:41 [DEBUG] Entering SendUserAuthBanner()
2019-07-31 22:23:41 [DEBUG] BP: paddingSz = 6
2019-07-31 22:23:41 [DEBUG] CreateMac hmac-sha2-256
2019-07-31 22:23:41 [DEBUG] Encrypt aes128-cbc
2019-07-31 22:23:41 [DEBUG] Entering wolfSSH_SendPacket()
2019-07-31 22:23:41 [DEBUG] Embed Send trying to send 144
2019-07-31 22:23:41 [DEBUG] Embed Send sent 144
2019-07-31 22:23:41 [DEBUG] SB: Shrinking output buffer
2019-07-31 22:23:41 [DEBUG] Entering ShrinkBuffer()
2019-07-31 22:23:41 [DEBUG] SB: usedSz = 0, forcedFree = 0
2019-07-31 22:23:41 [DEBUG] SB: releasing dynamic buffer
2019-07-31 22:23:41 [DEBUG] Leaving ShrinkBuffer()
2019-07-31 22:23:41 [DEBUG] Leaving SendUserAuthBanner()
2019-07-31 22:23:41 [DEBUG] accept state: ACCEPT_SERVER_USERAUTH_ACCEPT_SENT
2019-07-31 22:23:41 [DEBUG] PR1: size = 16

NOTE: It does not seem to happen with openssh ssh client.  I am trying to understand why there is an inconsistency.

JacobBarthelmeh, 2019-08-02:
HI csenanda, are you using non blocking with the example wolfSSH server?

I tried reproducing with `cd wolfssh && git checkout v1.4.0-stable && ./autogen.sh && ./configure && make && ./examples/echoserver/echoserver`

Using putty.exe version 0.71 `putty.exe -P 22222 jill@<the machines IP>` and could not reproduce it. What version of putty is being used and command for the connection? If it is with non blocking then I could see a possible change in behavior (-N with wolfSSH echoserver) due to the network that we would need to debug in wolfSSH.

Also may you tell me some more about the build and setup? Is it being built on Windows and how is wolfSSL being built before being linked.

Regards,
Jacob

csenanda, 2019-08-05:
Hello Jacob,

Thanks for your response.  I compiled and ran wolfssl/wolfssh on Ubuntu 14.04.6 LTS. Here is what I did.  
cd wolfssl
git checkout v1.4.0-stable
./autogen.sh && ./configure --enable-cryptonly --enable-ssh
make && make install
cd ../wolfssh
git checkout v1.4.0-stable
./autogen.sh && ./configure --enable-debug --enable-sftp
make && make install

I ran the echoserver as mentioned below and connected from putty/plink.exe v0.72 running on windows 10.  Attached debug logs from running the echoserver and plink ssh client trying to authenticate using username and public key.  I converted ssh-rsa key to putty private key format using puttgen before running plink with -i option.

Looking forward your help.  Thanks for your support.
Regards,
Nanda.

[plink_ssh_client.log](https://github.com/wolfSSL/wolfssh/files/3465986/plink_ssh_client.log)
[echoserver_debug.log](https://github.com/wolfSSL/wolfssh/files/3465941/echoserver_debug.log)
[echoserver_with_identity_key.log](https://github.com/wolfSSL/wolfssh/files/3465963/echoserver_with_identity_key.log)

csenanda, 2019-08-29:
Hello Jacob,

Did you get chance to review my earlier logs.  Do you think if I miss any configuration?
Your response will be very helpful.

Thanks,
nanda.

JacobBarthelmeh, 2019-10-24:
Hi @csenanda

Try with the most recent wolfSSH. Going back to wolfSSH version 1.4.0 we have made some updates for better interoperability with Putty since then. Notably a fix for AES CTR use and also a fix for password authentication, also Putty often seems to be using aes-ctr it may be worth turning on aes-ctr when building by configuring wolfSSL with --enable-aesctr.

Regards,
Jacob

csenanda, 2019-10-25:
Hello Jacob,

Thanks for the reply but it was too late a reply, we have decided not to move away from wolfssh as it was open for quite some time.

JacobBarthelmeh, 2019-10-25:
@csenanda thanks for feedback, if you come to a spot where wolfSSH is being re-evaluated in the future go ahead and contact us through support (at) wolfssl .com if any issues should arise

Regards,
Jacob

closed 2019-10-25 by JacobBarthelmeh