wolfSSL/wolfssh #188: wolfSSH_SFTP_SendReadPacket returns more bytes than asked for!?

bagder, 2019-08-16:
Continuing my efforts of using this from curl, I call wolfSSH_SFTP_SendReadPacket to read data from the remote file.

I have an 18 byte file on my OpenSSH sftp test server and I run this curl command line:

    ./src/curl -u [credentials] sftp://localhost/~/hello -v

It makes curl open and stat the file just fine, determining that it is indeed 18 bytes. It then moves on to read the file (see link to the read function below) which right now asserts because on the first call when asked to read 18 bytes - it even says to wolfssh that the output buffer is 18 bytes - it very often returns 29. I have not checked to see if this truly writes more data into the buffer or if it just returns the wrong value - or perhaps I just have misunderstood how to use it?

Appreciate your feedback!

https://github.com/curl/curl/blob/96d36c8707e3417fa881ff4f2cdb0277c65c7668/lib/wolfssh.c#L245-L281

~~~
!!!!!!!!! wsftp_recv, max 18 bytes from offset 0
!!!!!!!!! wsftp_recv returns 29
curl: wolfssh.c:273: wsftp_recv: Assertion `rc <= (int)len' failed.
Aborted

JacobBarthelmeh, 2019-08-16:
@bagder investigating into it

bagder, 2019-08-16:
The comment in the source code explains what wasn't obvious in the header file. That 'ofs' is a pointer to two 32 bit values. So my code passes in totally crazy input...
~~~
 * ofst     offset to start reading at. 2 word32 array with value at ofst[0]
 *          being the lower and ofst[1] being the upper.

bagder, 2019-08-16:
Nope, premature conclusion. I updated my code to pass in a pointer to an array with two `word32` values, but I still get 29 back.

JacobBarthelmeh, 2019-08-20:
I opened this PR (https://github.com/wolfSSL/wolfssh/pull/190) to start on the debugging path for this. Since blocking worked ok, it looks like has to do with the non blocking portion of the code. Am working on debugging the state machine for that API with non blocking enabled.

bagder, 2019-08-20:
Lovely. I'll be happy to run tests and poke from my end of things if you have any specific ideas or queries!

JacobBarthelmeh, 2019-08-21:
May I get the debug log from when it hits the issue and one from when it does not. This is with building wolfSSH with --enable-debug or the macro DEBUG_WOLFSSH and making a call to wolfSSH_Debugging_ON() to turn on debug print outs before the function in question get called.

bagder, 2019-08-22:
Sure. Here are two logs from just now with the debug output from the point where `wolfSSH_SFTP_Open()` returns successfully for the file.

[wolfssh-log-success.txt](https://github.com/wolfSSL/wolfssh/files/3528628/wolfssh-log-success.txt)
[wolfssh-log-fail.txt](https://github.com/wolfSSL/wolfssh/files/3528630/wolfssh-log-fail.txt)

JacobBarthelmeh, 2019-08-22:
Great! That clears it up. Thank you @bagder!

JacobBarthelmeh, 2019-08-22:
The most recent commit on PR #190 should resolve the issue now

bagder, 2019-08-22:
Lovely, I'll try it out tomorrow!

bagder, 2019-08-23:
Confirmed! I cannot reproduce the problem anymore with #190 applied. Thanks!

JacobBarthelmeh, 2019-08-23:
Thanks again for the report and for confirming the fix! Will go ahead and close out this issue.

closed 2019-08-23 by JacobBarthelmeh