wolfSSL/wolfssh #199: Build wolfssh on Windows to use sftp client

future1122, 2019-09-19:
Enviroment:
- [ Window10]
- [ VS 2015]
- [ wolfssh v1.4.2]

I want to build wolfssh to use sftpclient. But I  have problems when I build it 
`Error C1083 Can not open file: “wolfssl/wolfcrypt/settings.h”: No such file or directory	wolfssh	F:\library\wolfssh\wolfssh\ssh.h	30	
`
The error occurred.
Do I have to build wolf ssl first?
By the way , my real purpose is to integrate sftp client to my own application, Do you have any suggestions?
Thanks.

kaleb-himes, 2019-09-20:
Hi @future1122,

Yes you need to build wolfSSL first but the file it's trying to include and cannot find is not part of the build step so the include path for the project must not be able to find wolfSSL.

We provide a guide here for building wolfSSH and wolfSSL on windows: https://www.wolfssl.com/documentation/building-wolfssh-on-windows.pdf

Please let us know if you have any other questions.

Warm Regards,

K

ejohnstown, 2019-10-10:
We have recently made a few improvements to building wolfSSH on Windows. Did you see the README file in the wolfSSH directory ide/winvs?

ejohnstown, 2019-10-24:
Since I haven't heard from you for a while, I'm going to assume this is resolved. If you have any other questions, feel free to open an issue or send us email at support@wolfssl.com. Thank you.

closed 2019-10-24 by ejohnstown