wolfSSL/wolfssh #221: Compilation errors when compiling wolfSSH for 32 bit

guidovranken, 2019-11-29:
I cannot (cross)compile wolfSSH for 32 bit (x86).

I configure using ```setarch i386 ./configure --enable-static --enable-all``` and ```CFLAGS``` contains ```-m32```.

```
src/port.c:125:49: error: shift count >= width of type [-Werror,-Wshift-count-overflow]
            long offset = ((long)shortOffset[1] << 32) | shortOffset[0];
                                                ^  ~~
src/port.c:133:49: error: shift count >= width of type [-Werror,-Wshift-count-overflow]
            long offset = ((long)shortOffset[1] << 32) | shortOffset[0];
                                                ^  ~~
2 errors generated.
src/wolfscp.c:1568:25: error: implicit conversion loses integer precision: 'word64' (aka 'unsigned long long') to '__time_t' (aka 'long') [-Werror,-Wshorten-64-to-32]
        tmp[0].tv_sec = aTime;Makefile:1231: recipe for target 'src/src_libwolfssh_la-port.lo' failed
make[1]: *** [src/src_libwolfssh_la-port.lo] Error 1

make[1]: *** Waiting for unfinished jobs....
                      ~ ^~~~~
src/wolfscp.c:1570:25: error: implicit conversion loses integer precision: 'word64' (aka 'unsigned long long') to '__time_t' (aka 'long') [-Werror,-Wshorten-64-to-32]
        tmp[1].tv_sec = mTime;
                      ~ ^~~~~
2 errors generated.
Makefile:1252: recipe for target 'src/src_libwolfssh_la-wolfscp.lo' failed
make[1]: *** [src/src_libwolfssh_la-wolfscp.lo] Error 1
src/wolfsftp.c:3810:42: error: shift count >= width of type [-Werror,-Wshift-count-overflow]
    atr->sz[1] = (word32)((stats.st_size >> 32) & 0xFFFFFFFF);
                                         ^  ~~
src/wolfsftp.c:3852:42: error: shift count >= width of type [-Werror,-Wshift-count-overflow]
    atr->sz[1] = (word32)((stats.st_size >> 32) & 0xFFFFFFFF);
                                         ^  ~~
src/wolfsftp.c:7489:39: error: implicit conversion loses integer precision: 'word64' (aka 'unsigned long long') to 'long' [-Werror,-Wshorten-64-to-32]
                    WFSEEK(state->fl, offset, 0);
                    ~~~~~~~~~~~~~~~~~~^~~~~~~~~~
./wolfssh/port.h:221:42: note: expanded from macro 'WFSEEK'
    #define WFSEEK(s,o,w)     fseek((s),(o),(w))
                              ~~~~~      ^
3 errors generated.
```

JacobBarthelmeh, 2019-12-07:
Thanks for the report Guido, we got a fix for the compile time warning with off_t and one under way for the conversion loss warnings.

guidovranken, 2019-12-13:
Fixed, thanks.

closed 2019-12-13 by guidovranken