wolfSSL/wolfssh #233: how do I enable publickey auth in my SFTP client code?

bagder, 2020-01-15:
Hi again!

We've now merged initial support for wolfSSH in curl. It however struck me I haven't fixed publickey auth in the curl use of wolfssh (https://github.com/curl/curl/issues/4820), and I find the docs rather lacking in this aspect.

What do I need to do to enable publickey auth? In my test case the server accepts publickey auth but my callback set with `wolfSSH_SetUserAuth()` gets called numerous times but only with `authtype == WOLFSSH_USERAUTH_PASSWORD`:

https://github.com/curl/curl/blob/68403cdbc6325bcbe4e4128c263db5d04a3c77c6/lib/vssh/wolfssh.c#L393-L395

(A secondary issue is that wolfssh git master doesn't build for me after I updated this afternoon (thinking maybe I needed an update), which is surprising since your CI builds all look green but I'll save that for now.)

JacobBarthelmeh, 2020-01-16:
Hey Daniel,

That's great with the initial support merged in!

Thanks for the report, I see the curl test number in the link. I did a quick sanity check on the password based example bundled in wolfSSH and it went okay, so looks like I will need to dig into it a bit more to find the difference.

```
cd wolfssh
./examples/server/server
ssh -i ./keys/hansel-key-rsa.pem -p 22222 hansel@127.0.0.1
```

Regards,
Jacob

JacobBarthelmeh, 2020-01-23:
I opened a PR with changes I did for resolving linking while investigating this.

JacobBarthelmeh, 2020-01-24:
After more investigation this looks like something we need to change in wolfSSH. Dealing with how the flag for public key connections on the client side get handled. You can change this location of code (https://github.com/wolfSSL/wolfssh/blob/master/src/internal.c#L3813) for testing but we will work on a better runtime way for setting public key on the client side

closed 2020-07-07 by JacobBarthelmeh