wolfSSL/wolfssh #245: 32bit Linux built echoserver using ECC key segmentation fault

skhasan, 2020-02-22:
I have wolfSSH 1.4.3 and wolfSSL 4.3.0 in 32bit Linux machine and faced a seg fault issue with echoserver when using ECC key (-e argument). But this issue does not happen in 64bit Linux machine. I have also found that this issue is generated from wolfSSL's integer.c file. Can anyone help me with the fix?

JacobBarthelmeh, 2020-02-24:
Hi skhasan, thanks for reaching out to us! Can you walk me through the build steps taken? How was wolfSSL configured i.e. ./configure --enable-ssh, and how was wolfSSH configured? I gave this a quick try checking out those exact versions (1.4.3 and 4.3.0) on a 32bit Ubuntu VM using "./examples/echoserver/echoserver -E" and "./examples/echoserver/echoserver -e".

Also does the issue go away if building wolfSSL with "--enable-sp --enable-sp-math"? That tells wolfSSL to use a different math library.

skhasan, 2020-02-25:
Hi Jacob, Nice to meet you and thanks.
> How was wolfSSL configured i.e. ./configure --enable-ssh

That's right. Only "--enable-ssh" has been used.

> how was wolfSSH configured?

Default configuration. i.e. ./configure

> I gave this a quick try checking out those exact versions (1.4.3 and 4.3.0) on a 32bit Ubuntu VM

Did you able to reproduce the issue?

> Also does the issue go away if building wolfSSL with "--enable-sp --enable-sp-math"?

No. Instead the following build-error occurred -
***************************************************
src/internal.c: In function ‘wolfSSH_oct2dec’:
src/internal.c:8721:15: warning: implicit declaration of function ‘mp_todecimal’; did you mean ‘mp_tohex’? [-Wimplicit-function-declaration]
         ret = mp_todecimal(&tmp, decimalString);
               ^~~~~~~~~~~~
               mp_tohex
src/internal.c:8721:15: warning: nested extern declaration of ‘mp_todecimal’ [-Wnested-externs]
  CC       src/libwolfssh_la-log.lo
  CC       src/libwolfssh_la-io.lo
  CC       src/libwolfssh_la-port.lo
  CCLD     src/libwolfssh.la
  CCLD     examples/client/client
  CCLD     examples/server/server
/usr/bin/ld: src/.libs/libwolfssh.so: undefined reference to `mp_todecimal'
collect2: error: ld returned 1 exit status
**************************************

JacobBarthelmeh, 2020-02-25:
Thanks for giving the --enable-sp a try. I have not been able to reproduce it yet, with the default configure on one of the 32 bit systems at the office here. I will give it a try on another one that we have to see if it can be reproduced there.

Can you give us a stack trace of where it is failing? To do this I suggest compiling wolfSSL with "./configure --enable-ssh --enable-debug" and wolfSSH with "./configure --enable-static --disable-shared --enable-debug" then running "gdb ./examples/echoserver/echoserver -e".

skhasan, 2020-03-02:
Sorry for my late reply.
Here is the stack trace log:

(gdb) r
Starting program: /home/hasan/Downloads/wolfssh-1.4.3-stable/examples/echoserver/echoserver -e
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/i386-linux-gnu/libthread_db.so.1".
2020-03-03 00:04:24 [DEBUG] Entering wolfSSH_Init()
2020-03-03 00:04:24 [DEBUG] Leaving wolfSSH_Init(), returning 0
2020-03-03 00:04:24 [DEBUG] Entering wolfSSH_CTX_new()
2020-03-03 00:04:24 [DEBUG] Entering CtxInit()
2020-03-03 00:04:24 [DEBUG] Leaving wolfSSH_CTX_new(), ctx = 0x424750
2020-03-03 00:04:24 [DEBUG] Entering wolfSSH_CTX_SetBanner()
2020-03-03 00:04:24 [INFO]   setting banner to: "wolfSSH Example Echo Server
"
2020-03-03 00:04:24 [DEBUG] Entering wolfSSH_CTX_UsePrivateKey_buffer()
2020-03-03 00:04:24 [DEBUG] Entering wolfSSH_new()
2020-03-03 00:04:24 [DEBUG] Entering SshInit()
2020-03-03 00:04:24 [DEBUG] Entering HandshakeInfoNew()
2020-03-03 00:04:24 [DEBUG] Leaving wolfSSH_new(), ssh = 0x425840
2020-03-03 00:04:24 [DEBUG] Entering wolfSSH_SetHighwaterCtx()
2020-03-03 00:04:24 [DEBUG] Entering wolfSSH_SetHighwater()

2020-03-03 00:04:32 [DEBUG] Entering wolfSSH_set_fd()
2020-03-03 00:04:32 [DEBUG] Entering wolfSSH_accept()
2020-03-03 00:04:32 [DEBUG] SSH-2.0-wolfSSHv1.4.3

2020-03-03 00:04:32 [DEBUG] Entering wolfSSH_SendPacket()
2020-03-03 00:04:32 [DEBUG] Embed Send trying to send 23
2020-03-03 00:04:32 [DEBUG] Embed Send sent 23
2020-03-03 00:04:32 [DEBUG] SB: Shrinking output buffer
2020-03-03 00:04:32 [DEBUG] Entering ShrinkBuffer()
2020-03-03 00:04:32 [DEBUG] SB: usedSz = 0, forcedFree = 0
2020-03-03 00:04:32 [DEBUG] SB: releasing dynamic buffer
2020-03-03 00:04:32 [DEBUG] Leaving ShrinkBuffer()
2020-03-03 00:04:32 [DEBUG] accept state: SERVER_VERSION_SENT
2020-03-03 00:04:32 [DEBUG] Receive: recvd = 41
2020-03-03 00:04:32 [DEBUG] Entering ShrinkBuffer()
2020-03-03 00:04:32 [DEBUG] SB: usedSz = 0, forcedFree = 0
2020-03-03 00:04:32 [DEBUG] SB: releasing dynamic buffer
2020-03-03 00:04:32 [DEBUG] Leaving ShrinkBuffer()
2020-03-03 00:04:32 [DEBUG] accept state: CLIENT_VERSION_DONE
2020-03-03 00:04:32 [DEBUG] Entering SendKexInit()
2020-03-03 00:04:32 [DEBUG] BP: paddingSz = 6
2020-03-03 00:04:32 [DEBUG] CreateMac none
2020-03-03 00:04:32 [DEBUG] Encrypt none
2020-03-03 00:04:32 [DEBUG] Entering wolfSSH_SendPacket()
2020-03-03 00:04:32 [DEBUG] Embed Send trying to send 336
2020-03-03 00:04:32 [DEBUG] Embed Send sent 336
2020-03-03 00:04:32 [DEBUG] SB: Shrinking output buffer
2020-03-03 00:04:32 [DEBUG] Entering ShrinkBuffer()
2020-03-03 00:04:32 [DEBUG] SB: usedSz = 0, forcedFree = 0
2020-03-03 00:04:32 [DEBUG] SB: releasing dynamic buffer
2020-03-03 00:04:32 [DEBUG] Leaving ShrinkBuffer()
2020-03-03 00:04:32 [DEBUG] Leaving SendKexInit(), ret = 0
2020-03-03 00:04:32 [DEBUG] accept state: SERVER_KEXINIT_SENT
2020-03-03 00:04:32 [DEBUG] PR1: size = 8
2020-03-03 00:04:32 [DEBUG] Receive: recvd = 8
2020-03-03 00:04:32 [DEBUG] Decrypt none
2020-03-03 00:04:32 [DEBUG] PR2: size = 1392
2020-03-03 00:04:32 [DEBUG] Receive: recvd = 1384
2020-03-03 00:04:32 [DEBUG] Decrypt none
2020-03-03 00:04:32 [DEBUG] VerifyMac none
2020-03-03 00:04:32 [DEBUG] VM: inSz = 1392
2020-03-03 00:04:32 [DEBUG] VM: seq = 0
2020-03-03 00:04:32 [DEBUG] VM: keyLen = 0
2020-03-03 00:04:32 [DEBUG] DoPacket sequence number: 0
2020-03-03 00:04:32 [DEBUG] Decoding MSGID_KEXINIT
2020-03-03 00:04:32 [DEBUG] Entering DoKexInit()
2020-03-03 00:04:32 [DEBUG] DKI: KEX Algorithms
[... 210 lines cut ...]
2020-03-03 00:04:32 [DEBUG] Leaving DoUserAuthRequest(), ret = 0
2020-03-03 00:04:32 [DEBUG] PR3: peerMacSz = 16
2020-03-03 00:04:32 [DEBUG] PR4: Shrinking input buffer
2020-03-03 00:04:32 [DEBUG] Entering ShrinkBuffer()
2020-03-03 00:04:32 [DEBUG] SB: usedSz = 0, forcedFree = 1
2020-03-03 00:04:32 [DEBUG] SB: releasing dynamic buffer
2020-03-03 00:04:32 [DEBUG] Leaving ShrinkBuffer()
2020-03-03 00:04:32 [DEBUG] PR5: txCount = 112, rxCount = 80
2020-03-03 00:04:32 [DEBUG] PR1: size = 16
2020-03-03 00:04:32 [DEBUG] Receive: recvd = 16
2020-03-03 00:04:32 [DEBUG] PR2: size = 308
2020-03-03 00:04:32 [DEBUG] Receive: recvd = 292
2020-03-03 00:04:32 [DEBUG] DecryptAead aes128-gcm@openssh.com
2020-03-03 00:04:32 [DEBUG] DoPacket sequence number: 5
2020-03-03 00:04:32 [DEBUG] Decoding MSGID_USERAUTH_REQUEST
2020-03-03 00:04:32 [DEBUG] Entering DoUserAuthRequest()
2020-03-03 00:04:32 [DEBUG] Entering DoUserAuthRequestPublicKey()
2020-03-03 00:04:32 [DEBUG] DUARPK: Calling the userauth callback
2020-03-03 00:04:32 [DEBUG] DUARPK: callback result = 0
2020-03-03 00:04:32 [DEBUG] Entering DoUserAuthRequestEcc()
realloc(): invalid pointer

Program received signal SIGABRT, Aborted.
0xb7fd4d61 in __kernel_vsyscall ()
(gdb)

skhasan, 2020-03-03:
Sorry, this issue is still in opened state.

JacobBarthelmeh, 2020-03-03:
Thanks, the stack trace helps. I have a suspicion of what is happening now. Am working on verifying it and a code change if needed.

JacobBarthelmeh, 2020-03-03:
@skhasan I opened a PR that resolved the issue when testing locally.

skhasan, 2020-03-04:
Thank you so much for the fix.
I confirm that echoserver is now running on 32bit Linux with ECC key.

closed 2020-03-05 by JacobBarthelmeh