wolfSSL/wolfssh #272: wolfSSH reverse port forwarding

pavankumarka, 2020-07-30:
Hi WolfSSH Team,
I am a newbie here, I see WolfSSH server, client and Client port forwarding examples provided as separate features/examples.  In the port forwarding example there is expectation of  **openSSH server in port forwarding mode** to be started in background. 
Comments: Learning and adopting WolfSSH purpose will be diverted, if expecting different sources (OpenSSH) in my view. 

Requirement: Is there an example or approach explained which I can refer for "SSH Server/echoServer example build with reserve/remote port forwarding feature" ?

Best regards,
Pavan

ejohnstown, 2020-08-26:
We only have the client side behavior for port forwarding, not the server side, yet. It is a feature we can add to our server, though. Can I put you in touch with a sales manager?

ejohnstown, 2021-07-20:
I'm going to close this issue. We are currently working on it as a project. See PR #343.

closed 2021-07-20 by ejohnstown