wolfSSL/wolfssh #302: Certificate-based authentication

execretor, 2020-12-23:
Hi, 

wolfSSH is developed with the devices world (such as IoT or embedded devices) as described in the home page.
But, using a distributed echosystem makes very difficoult to managed standard authentication modes such as passwords or public/private key couples.

For example, in a fleet of thousends devices, every ssh user needs to store its public key in every device and it isn't possible in a industrial environment. Managing permissions and secure access easily become an hell.

In the SSH world there are two different specifications that tries to solve the problem:
- RFC 6187 - X.509v3 Certificates for Secure Shell Authentication (March 2011)
- http://cvsweb.openbsd.org/cgi-bin/cvsweb/~checkout~/src/usr.bin/ssh/PROTOCOL.certkeys?content-type=text/plain

The first one is a standard related to SSHv2 protocol, the second one is an extension proposed and fully supported by OpenSSH. 

Is the support of these two specifications (or only one of them) planned for wolfssh?

Thanks,
Lorenzo

ejohnstown, 2020-12-23:
The echoserver is an example server. Its original goal was to return keypresses in the SSH stream back to the client. We happen to bolt demonstrations of new features added to the library onto it.

We do have x.509v3 certificate support on our roadmap for wolfSSH. Would you be interested in funding its development?
: ]

ejohnstown, 2022-05-19:
Lorenzo:

X.509 certificate support is a project that is in process and should be merged in in the next week or two. Thank you for the suggestion.

closed 2022-05-19 by ejohnstown