wolfSSL/wolfssh #311: WolfSSH 1.4.5 echoserver SFTP Error [-1001] "Channel data received"

fijam, 2021-01-25:
This might well be a problem between keyboard and chair but here it goes:

Server A: 
OpenSSH_8.4p1, OpenSSL 1.1.1i  8 Dec 2020 Linux 5.4.89-gentoo

Server B:
Linux 5.10.7-0-lts wolfSSL/wolfssl#1-Alpine
WolfSSL v4.6.0 ./configure --enable-aesctr --enable-aesni --enable-afalg --enable-ssh --enable-keygen --enable-opensslcoexist
WolfSSH v1.4.5 ./configure --enable-scp --enable-sftp --enable-keygen --enable-shell
grep -F "model name" /proc/cpuinfo model name      : AMD GX-212ZC SOC with Radeon(TM) R1E Graphics
gcc version 10.2.1 Target: x86_64-alpine-linux-musl

on Server B:
```
# ./examples/echoserver/echoserver
```

on Server A:

```
# sftp -P 22222 user@192.168.0.2
wolfSSH Example Echo Server
user@192.168.0.2's password:
Connected to 192.168.0.2.
sftp> put file.dat file.dat
Uploading file.dat to /tmp/./file.dat
file.dat                                                                               0%    0     0.0KB/s   --:-- ETA
client_loop: send disconnect: Broken pipe
Couldn't send packet: Connection reset by peer
Connection closed.
```

on Server B:

```
Error with SSH shutdown.
Error [-1001] "Channel data received" with handling connection.
```

The error depends on file size. I tested files with sizes of 1KB, 10KB, 100KB, 1MB and 10MB.
1KB and 10KB files succeed.
100KB file uploads 96% and stalls.
1MB file  sometimes succeeds, sometimes fails with Error [-1001] "Channel data received" with handling connection.
10MB file fails with  Error [-1001] "Channel data received" or "verify mac error" with handling connection

I also tested with a Windows machine running WinSCP 5.17.9 and got Error [-1001] "Channel data received" as well.

Please let me know if you need more details.

embhorn, 2021-01-25:
Hello @fijam 

Thanks for contacting wolfSSL Support. I've transferred this issue report to the wolfSSH project. I'll request an engineer to review and provide an update.

ejohnstown, 2021-01-26:
Which commit of wolfSSH are you using?

fijam, 2021-01-26:
https://github.com/wolfSSL/wolfssh/tree/v1.4.5-stable

ejohnstown, 2022-05-19:
Since you submitted this, we have addressed many reliability issues with wolfSFTP. I believe the issues you were seeing should be resolved at this time.

closed 2022-05-19 by ejohnstown