wolfSSL/wolfssh #341: libwolfssh not detected during configure for cross-compiling

misaalanshori, 2021-06-10:
I am trying to cross-compile wolfssh for arm, but it is not detecting libwolfssh that I have compiled. below is what I am doing:
```
❯ ./configure --host=arm-linux-gnueabi --prefix=/usr/arm-linux-gnueabi --enable-static LDFLAGS=-static --disable-stripping --without-manpages --without-tack --without-tests
......
checking termios.h usability... yes
checking termios.h presence... yes
checking for termios.h... yes
checking for socket in -lnetwork... no
checking for forkpty in -lutil... yes
checking for wolfCrypt_Init in -lwolfssl... no
configure: error: libwolfssl is required for wolfssh. It can be obtained from https://www.wolfssl.com/download.html/ .
❯ ls /usr/arm-linux-gnueabi/lib/libwolf*
/usr/arm-linux-gnueabi/lib/libwolfssl.a   /usr/arm-linux-gnueabi/lib/libwolfssl.so     /usr/arm-linux-gnueabi/lib/libwolfssl.so.24.4.0
/usr/arm-linux-gnueabi/lib/libwolfssl.la  /usr/arm-linux-gnueabi/lib/libwolfssl.so.24
```
The lib files are in the correct folder, but it is not getting detected. Where is the configure script looking for this file?

embhorn, 2021-06-10:
Hello @misaalanshori 

The configure line for wolfSSH needs to know where to look for the wolfssl library. 
Please try adding:
CFLAGS="-I/usr/arm-linux-gnueabi/include" LDFLAGS="-L/usr/arm-linux-gnueabi/lib"

Thanks,
wolfSSL Support

misaalanshori, 2021-06-10:
Okay thanks!

embhorn, 2021-06-14:
Hi @misaalanshori 

Do you have further questions? Can I mark this issue as closed?

misaalanshori, 2021-06-15:
I have not tested it as I'm currently busy. but I think it's safe to close this.

closed 2021-06-15 by misaalanshori