wolfSSL/wolfssh #362: wolfSSH on STM32L4

maborghe, 2021-08-31:
Hi everyone,
I'm trying to use wolfSSH on STM32L4 + FreeRTOS. Is it possible to do it and, if so, what are the steps to achieve that?
I see there is a dependency on wolfCrypt, which I guess I can solve as explained in #219 . What are the dependencies as regards the networking part? So far I've had some problems finding necessary include files such as  `netdb.h` (which I've solved using lwip, although I'm not sure it really works) and `sys/ioctl.h` (which I couldn't find anywhere).

Thanks in advance for your help

dgarske, 2021-08-31:
Hi @maborghe ,

For the wolfCrypt you will need `WOLFCRYPT_ONLY` and `WOLFSSL_PUBLIC_MP`. What network stack are you using? If you are using LWIP you can define `WOLFSSH_LWIP`. If you would like to implement your own networking layer you can define `WOLFSSH_USER_IO` and set your own read/write callbacks using `wolfSSH_SetIORecv` and `wolfSSH_SetIOSend`.

Let us know if you see any other issues.

Thanks,
David Garske, wolfSSL

dgarske, 2022-05-31:
Hi @maborghe ,

Some time has passed with no update, so I will go ahead and mark this issue closed. If you are still having any issues with wolfSSH on FreeRTOS/LWIP please feel free to re-open.

Thanks,
David Garske, wolfSSL

closed 2022-05-31 by dgarske