wolfSSL/wolfssh #363: Uninitialized Variables in internal.c  port.h  log.h

gojimmypi, 2021-09-06:
I'm in the process of porting wolfssh to the Espressif ESP-IDF for inclusion in [ide](https://github.com/wolfSSL/wolfssh/tree/master/ide) directory, similar to that for the [wolfssl IDE](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF). And yes, note the [wolfssl IDE](https://github.com/wolfSSL/wolfssl/tree/master/IDE) directory is uppercase, whereas the [wolfssh ide](https://github.com/wolfSSL/wolfssh/tree/master/ide) directory is lower case.

The Espressif `idf.py build` treats all warnings as errors. That's probably a good idea to help keep code clean and reliable.

As an interim workaround, I added this to my copy of `internal.c` in the esp-idf:

```
#pragma GCC diagnostic ignored "-Wmaybe-uninitialized"
```

This issue is just a reminder that these warnings exist and should eventually be cleaned up.

For reference, here's the result of `idf.py build | grep -i error` for a sample project showing the uninitialized variables:

```
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:1001:19: error: 'hashId' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:1002:45: error: 'sK' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:997:45: error: 'cK' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:8974:13: error: 'channelTypeSz' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/wolfssh/port.h:380:31: error: 'channelType' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:2412:22: error: 'side' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:3129:23: error: 'e' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:8703:13: error: 'serviceNameSz' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/wolfssh/port.h:380:31: error: 'serviceName' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:8708:13: error: 'authNameSz' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/wolfssh/port.h:380:31: error: 'authName' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:3942:31: error: 'pw' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:4467:27: error: 'digestSz' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:4467:27: error: 'pk' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:4860:15: error: 'newChannel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:9133:9: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:9182:9: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:9231:9: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:9294:28: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:9341:24: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/wolfssh/log.h:76:27: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
/home/gojimmypi/esp/esp-idf/components/wolfssh/src/internal.c:9873:9: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
```

ejohnstown, 2021-09-09:
The funny thing about this is we set `-Wmaybe-uninitialized` in configure. When we build the code on macOS and in Linux, using both clang and gcc, none are reporting the warning. I went through a few of these cases and the variables aren't set in error conditions and aren't referenced once the error is detected.

I'm going to get ESP-IDF up and running here. I think we have some proto boards in the office. It'll be awesome to see wolfSSH running on it.

gojimmypi, 2021-09-10:
Thanks for taking a look at this. I have a _very_ preliminary [wolfssl-esp32](https://github.com/gojimmypi/wolfssl-esp32) repo, with the [ESP32 SSH Server Project](https://github.com/gojimmypi/wolfssl-esp32/tree/main/wolfssl_ssh_server) that I've been working on. (just a hobby; yes, it is a quite an ugly a hack at the moment)

I have a copy of my [wolfSSH ESP IDF Component](https://github.com/gojimmypi/wolfssl-esp32/tree/main/wolfssl_ssh_server/esp-idf-component), but note it gets moved to the root directory in my local, not-yet-pushed repo changes. I have a real mess of a TCP listener at the moment with the dreaded "guru meditation panic" when a packet is received.

I have a script that will push files to the IDF, similar to the [SSL Setup for Espressif](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/setup.sh), but had [problems with GitHooks](https://twitter.com/gojimmypi/status/1434912665822924810?s=20) in Visual Studio trying to push that to GitHub. Yes, a bit odd, perhaps: I'm writing code in Visual Studio, and using WSL to compile and write firmware to the ESP32.

The plan for this weekend is to revisit this, push all the changes I have, and ask for help on the TCP Listener. I had a suggestion from Jacob today that it might be a misaligned SSL/SSH configuration. That sounds very likely as [I've struggled](https://github.com/espressif/esp-wolfssl/issues/11#issuecomment-908805153) to learn how the config works.

ejohnstown, 2022-05-19:
I think we've taken care of these items. I'm going to close this issue.

closed 2022-05-19 by ejohnstown