wolfSSL/wolfssh #366: Question about CPU footprint

docent-net, 2021-09-27:
Hi, posting this question as an issue here, as it might be a good reference for others.

We're using classic openssh server in our infrastructure. However, we're having issues with high CPU demands when even low number of concurrent users are connected (5-10). That's because sshd is single - threaded and encryption takes a lot of CPU.

The only way for scaling this is (we think) scaling vertically by adding more CPUs and using CPU pinning. But this is really cumbersome as we're in k8s in some public cloud.

So, we're looking for another sshd implementation that might be better performing or maybe could be configured in a more performant way. And here we are analyzing wolfssh.

We found this article about memory footprint of wolfssl - and that looks very promising! https://www.wolfssl.com/wolfssh-sftp-performance/ However not word about CPU utilization there. According to this, https://www.wolfssl.com/docs/tls13/ TLS 1.3 will be probably more performant and less CPU intensive.

Could you please elaborate a little about how WolfSSH/WolfSSL compares to openssh in terms of cpu utilization? And how does scalability look like? As far as I understand from documentation, WolfSSL is multi-threaded (https://www.wolfssl.com/docs/wolfssl-manual/ch2/), with optional SINGLE_THREADED switch. However, not sure if this multithreading helps a lot from a performance perspective, as there is only one mutex (used for the session cache)?

Thank you

ejohnstown, 2021-09-28:
I can't comment on OpenSSH's CPU utilization. I've peeked at their code in the past, and they are spinning up multiple threads for handling the peer socket and one per channel, connected with pipes.

wolfSSL and wolfSSH are two separate libraries. wolfSSH uses the wolfCrypt portion of wolfSSL. wolfSSL's SINGLE_THREADED option is for TLS operation. You would normally run each TLS session in its own thread. There is some shared state, the session cache, that needs mutexes. Single threaded removes the mutexes.

In wolfSSL, we also have asynchronous cryptography usage. When doing some crypto operations, the session will return a WANT_WRITE-like code meaning the session is waiting for the crypto hardware. wolfSSH does not take advantage of this yet.

Cryptography is always going to be CPU intensive. wolfCrypt can offload crypto operations to additional hardware. For example, wolfCrypt can take advantage of AES-NI, AVX1/2, etc on the Intel processors. We can always add more as part of a support package. We have a single-precision math library that performs public key cryptography and runs super fast in constant time branch-free, but the code is somewhat large. There's always trade-offs. wolfSSH is using the same cryptography as wolfSSL, and is using many of the same algorithms. (ECDSA, ECDHE, AES-GCM, etc.)

wolfSSH's server is still a work in progress. (There's a reason it is in the examples directory.) It is missing authentication, but wolfSSH expects the user to add it with an authentication callback. (It can run on anything including systems without users.)

Can I put you in touch with one of our sales managers to discuss some consulting? We can help with porting and integration.

ejohnstown, 2022-05-20:
Do you have any other questions, or did I misunderstand what you asked? Would you like to follow up?

closed 2022-05-27 by ejohnstown