wolfSSL/wolfssh #382: CMake Support

Shadorain, 2022-01-14:
Hello! I was wondering why WolfSSH does not have support for CMake (a `CMakeLists.txt`) when WolfSSL does.  
I am trying to build WolfSSH and WolfSSL as libraries for an embedded project and I am using CMake, but can only compile SSL because it is setup for CMake and SSH is not.
I'm just wondering the direction here, was there ever supposed to be CMake support for WolfSSH.
Thank you!

ejohnstown, 2022-05-19:
We haven't gotten around to it. There hasn't been much call for it yet. It is on the roadmap.

ejohnstown, 2024-01-12:
I have started digging into this, but it will take a little while as I have several things I'm working on at the same time. I'm going to close this issue.

closed 2024-01-12 by ejohnstown