wolfSSL/wolfssh #402: internal.c error: variable may be used uninitialized in this function

gojimmypi, 2022-04-06:
I'm compiling the `wolfssh` using the Espressif ESP-IDF. The default compiler settings are such that possibly uninitialized variable warnings are treated as errors. It is probably a good idea not to just ignore the warnings.

I've been able to appease the compiler by simply assigning an initial value of zero. I could submit a PR that does this. Alternatively perhaps the logic needs to be more closely investigated?

```
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssl/wolfssl/wolfcrypt/settings.h:2348:14: warning: #warning "For timing resistance / side-channel attack prevention consider using harden options" [-Wcpp]
             #warning "For timing resistance / side-channel attack prevention consider using harden options"
              ^~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'GenerateKeys':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:997:49: error: 'sK' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                                   sK->macKey, sK->macKeySz,
                                               ~~^~~~~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:992:45: error: 'cK' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                               cK->macKey, cK->macKeySz,
                                           ~~^~~~~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'SendChannelOpen':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:9314:13: error: 'channelTypeSz' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         idx += channelTypeSz;
         ~~~~^~~~~~~~~~~~~~~~
In file included from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/wolfssh/ssh.h:39,
                 from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:32:
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/wolfssh/port.h:384:31: error: 'channelType' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     #define WMEMCPY(d,s,l)    memcpy((d),(s),(l))
                               ^~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:9287:17: note: 'channelType' was declared here
     const char* channelType;
                 ^~~~~~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'DoKexDhReply':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:3181:23: error: 'e' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                 ret = wc_RsaPublicKeyDecodeRaw(n, nSz, e, eSz,
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                &sigKeyBlock_ptr->sk.rsa.key);
                                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:49:
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'SendKexDhReply':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/misc.c:87:10: error: 'sigBlockSz' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     c[3] =  u32 & 0xff;
     ~~~~~^~~~~~~~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:6856:12: note: 'sigBlockSz' was declared here
     word32 sigBlockSz;
            ^~~~~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'SendUserAuthRequest':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:8999:13: error: 'serviceNameSz' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         idx += serviceNameSz;
             ^~
In file included from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/wolfssh/ssh.h:39,
                 from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:32:
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/wolfssh/port.h:384:31: error: 'serviceName' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     #define WMEMCPY(d,s,l)    memcpy((d),(s),(l))
                               ^~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:8890:17: note: 'serviceName' was declared here
     const char* serviceName;
                 ^~~~~~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:9004:13: error: 'authNameSz' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         idx += authNameSz;
             ^~
In file included from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/wolfssh/ssh.h:39,
                 from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:32:
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/wolfssh/port.h:384:31: error: 'authName' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     #define WMEMCPY(d,s,l)    memcpy((d),(s),(l))
                               ^~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:8888:17: note: 'authName' was declared here
     const char* authName;
                 ^~~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'DoUserAuthRequestPassword':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:4011:31: error: 'pw' may be used uninitialized in this function [-Werror=maybe-uninitialized]
             pw->newPasswordSz = 0;
             ~~~~~~~~~~~~~~~~~~^~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'DoUserAuthRequestPublicKey':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:4562:27: error: 'pk' may be used uninitialized in this function [-Werror=maybe-uninitialized]
[... 24 lines cut ...]
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'SendChannelData':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:9692:24: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         word32 bound = min(channel->peerWindowSz, channel->peerMaxPacketSz);
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/wolfssh/port.h:34,
                 from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/wolfssh/ssh.h:39,
                 from C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:32:
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'SendChannelRequest':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/wolfssh/log.h:76:27: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                           wolfSSH_Log(__VA_ARGS__); \
                           ^~~~~~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:9803:22: note: 'channel' was declared here
     WOLFSSH_CHANNEL* channel;
                      ^~~~~~~
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c: In function 'SendChannelSuccess':
C:/Users/gojimmypi/Desktop/esp-idf/components/wolfssh/src/internal.c:10224:9: error: 'channel' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         c32toa(channel->peerChannel, output + idx);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
[1096/1162] Building C object esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/falcon.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

cc: @dgarske

ejohnstown, 2022-04-07:
These weren't warnings elsewhere. Thanks for catching them!

ejohnstown, 2022-05-19:
Thank you for the updates. I'm going to close this issue at this time.

closed 2022-05-19 by ejohnstown