wolfSSL/wolfssh #406: wolfssh-1.4.8-stable dont build: syntax error near unexpected token `2.2'

BlackMage2, 2022-04-23:
i have CentOS Linux release 7.9.2009 (Core)
i get ever
```
./configure: line 3275: syntax error near unexpected token `2.2'
./configure: line 3275: `LT_PREREQ(2.2)'
```
wolfssl is already installed in /usr/local
i have aclocal 1.14.1

embhorn, 2022-04-25:
Hi @BlackMage2 

How was wolfSSL installed in /usr/local if the configure step is failing?
> wolfssl is already installed in /usr/local

Could you please share the config log?

embhorn, 2022-04-25:
Just covering all the basics here... Did you run `./autogen.sh` first?

BlackMage2, 2022-04-25:
yes that outputs:
`autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
Makefile.am:8: error: Libtool library used but 'LIBTOOL' is undefined
Makefile.am:8:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
Makefile.am:8:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
Makefile.am:8:   If 'LT_INIT' is in 'configure.ac', make sure
Makefile.am:8:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1`

embhorn, 2022-04-25:
And I know you provided the libtool version, but the first hit for that error is:
https://stackoverflow.com/questions/18978252/error-libtool-library-used-but-libtool-is-undefined

Could you please try running `libtoolize` then rerun the autogen script?
https://stackoverflow.com/a/18980043

BlackMage2, 2022-04-25:
THANKS, that was it
but compiling of wolfssh fails:

> autoreconf: Entering directory `.'
> autoreconf: configure.ac: not using Gettext
> autoreconf: running: aclocal --force -I m4
> autoreconf: configure.ac: tracing
> autoreconf: configure.ac: not using [...]

embhorn, 2022-04-25:
Did you build wolfSSL configured with `--enable-ssh`? Otherwise could you please share the configure command used for wolfSSL?
https://github.com/wolfSSL/wolfssh/blob/master/README.md#dependencies

BlackMage2, 2022-04-26:
wolfSSL configured with _./configure --enable-libssh2 --enable-openssh --enable-fwd --enable-echo --enable-shell_ and wolfSSH with _./configure --with-wolfssl=/usr/local --enable-scp --enable-fwd --enable-sftp --enable-shell --enable-keygen_

BlackMage2, 2022-04-26:
okay now i have successfully compiled both: wolfssl with `./configure --enable-libssh2 --enable-openssh --enable-ssh` and wolfssh with `./configure --with-wolfssl=/usr/local --enable-scp --enable-fwd --enable-sftp --enable-shell --enable-keygen`
a bit complicated to find out the right options

BlackMage2, 2022-04-26:
but no SSH server itself included?

embhorn, 2022-04-26:
> okay now i have successfully compiled both: wolfssl with `./configure --enable-libssh2 --enable-openssh --enable-ssh` and wolfssh with `./configure --with-wolfssl=/usr/local --enable-scp --enable-fwd --enable-sftp --enable-shell --enable-keygen` a bit complicated to find out the right options(i had enable-libssh2 and enable-shell and enable-openssh but no --enable-ssh xD)

In the wolfSSL config you only need `--enable-ssh` for wolfSSH. Those other options are configs for using wolfSSL with different SSH libraries.

> but no SSH server itself included?

The wolfSSH library is not a server, but it contains the parts needed to build one. There is a demonstration in the Echo Server example:
https://github.com/wolfSSL/wolfssh/blob/master/README.md#echoserver

>i need to compile dropbear or openssh with wolfssh lib?
how i can make that?

No, those projects contain their own implementations of the SSH protocol. They could be built to link against wolfSSL in order to use the cryptography provided by wolfCrypt (which is the purpose of those wolfSSL config options mentioned earlier).

BlackMage2, 2022-04-26:
wolfCrypt is a additional lib?

embhorn, 2022-04-26:
> wolfCrypt is a additional lib?

No, it is the cryptographic component of the wolfSSL library. wolfSSL also contains a full TLS/SSL stack, but part that is not used by wolfSSH.

embhorn, 2022-04-26:
You can actually use the wolfSSH echoserver example with the shell! I recommend testing with the echoserver with shell enabled and see if you actually need more than that. You'll want to update the accounts in wsUserAuth:

> Keep in mind, the echoserver has several fake accounts in its wsUserAuth() callback function. (jack, jill, hansel, and gretel) When the shell support is enabled, those fake accounts will not work. They don't exist in the system's passwd file. The users will authenticate, but the server will err out because they don't exist in the system. You can add your own username to the password or public key list in the echoserver. That account will be logged into a shell started by the echoserver with the privileges of the user running echoserver.

https://github.com/wolfSSL/wolfssh/blob/master/README.md#testing-notes

BlackMage2, 2022-05-02:
where is specified the port? should be 22222 but i didn´t find it

BlackMage2, 2022-05-02:
and on my tests is the echoserver not reachable, is should be reachable by the ssh client?

embhorn, 2022-05-02:
> where is specified the port? should be 22222 but i didn´t find it

Yes, the default port is 22222.

> and on my tests is the echoserver not reachable, is should be reachable by the ssh client?

I just tested successfully in linux using the instructions from:
https://github.com/wolfSSL/wolfssh/blob/master/README.md#examples

Please email support@wolfssl.com for priority support.

BlackMage2, 2022-05-02:
is the priority support free?

embhorn, 2022-05-02:
Yes, we do offer free pre-sales support!

BlackMage2, 2022-05-06:
is Linux version 2.6.32-042stab127.2 supported?

embhorn, 2022-05-09:
I would think so. Are you seeing issues? Please share any details.

ejohnstown, 2022-05-19:
On an older Linux distribution, the kernel shouldn't be a problem. You might get some compile errors due to the age of the compiler. Over the years different options are added to compilers and some are picked and built in differently between distributions. Are you having any specific issues? Or may I close this issue at this time?

closed 2022-05-27 by ejohnstown