wolfSSL/wolfssh #478: Windows 10/11 (x64) binaries?

lonnietc, 2022-12-19:
Hello,

I am interested in possibly using your WolfSSH components and in particular your SSH server in a project.

At this current stage in the development, I would just like to run a pre-compiled ssh server so that I can test out first before digging into compiling everything on Visual Studio 2019, or 2022.

The release do not show any binaries and I also cannot locate them on the website.

Where can I find them?
Thanks

embhorn, 2022-12-19:
Hello @lonnietc 

Thanks for contacting wolfSSL Support. We do not release precompiled binaries. Building the libraries is very straightforward, here are the instructions for Visual Studio:
https://github.com/wolfSSL/wolfssh/tree/master/ide/winvs

Please let us know if there are questions.

Can you tell us more about your project? Feel free to email facts@wolfssl.com if you'd prefer not to share publicly.

Kind regards,
@embhorn - wolfSSL Support

lonnietc, 2022-12-19:
Hello,

[building-wolfssh-on-windows.pdf](https://github.com/wolfSSL/wolfssh/files/10260840/building-wolfssh-on-windows.pdf)

Thanks for the quick reply. I actually think that I was able to build things according to the visual studio document for the wolfssh but it seemed to only build a library and I did not see and ssh executable which is what I was hoping to test.

closed 2022-12-20 by lonnietc