wolfSSL/wolfssh #511: error: Libtool library used but 'LIBTOOL' is undefined

marzban2030, 2023-04-23:
[image]

According to above picture for beginning:
1- I cloned the repository
2- Directory changed to repository folder (wolfssh)
3- I ran `./autogen.sh` command according to Build instruction but encouraged with an error:

Making missing build-aux directory.
Touching missing build-aux/config.rpath file.
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:38: installing 'build-aux/compile'
configure.ac:12: installing 'build-aux/config.guess'
configure.ac:12: installing 'build-aux/config.sub'
configure.ac:15: installing 'build-aux/install-sh'
configure.ac:15: installing 'build-aux/missing'
Makefile.am:8: error: Libtool library used but 'LIBTOOL' is undefined
Makefile.am:8:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
Makefile.am:8:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
Makefile.am:8:   If 'LT_INIT' is in 'configure.ac', make sure
Makefile.am:8:   its definition is in aclocal's search path.
Makefile.am: installing 'build-aux/depcomp'
parallel-tests: installing 'build-aux/test-driver'
autoreconf: automake failed exit status: 1

I want to build executable binary for armv5tej device inorder to remote port forwarding using SSH. Please help me!

kareem-wolfssl, 2023-04-24:
You will need libtool installed on your system to use ./configure, please confirm you have it installed.
If libtool is not available, you can use user_settings.h as an alternative.

marzban2030, 2023-04-24:
It solved the problem, Thank you.

kareem-wolfssl, 2023-04-24:
Glad to hear it helped!