wolfSSL/wolfssh #518: wolfssh on STM32U5

jandriea, 2023-05-18:
Hello,
I am trying to use WolfSSH on STM32U575 + MicroPython. Currently, I am using mbedTLS in MicroPython and I am wondering if I can just use the WolfSSH on top of it. Is it possible to do that? if so, what are the steps to achieve that?

embhorn, 2023-05-18:
Hi @jandriea 

Thanks for contacting wolfSSL Support. We are currently working on a wolfSSL implementation of MicroPython. Would you be interested in reviewing when it is ready?

wolfSSH is tightly integrated with the wolfCrypt library. It might be possible to use with mbedTLS, but not something we would be interested in supporting. That said, one of our engineers started doing some preliminary work on mapping MBEDTLS calls to wolfSSL calls, you can review those alpha changes here:
https://github.com/wolfSSL/wolfssl/compare/master...dgarske:mbedtls

Thanks,
@embhorn - wolfSSL Support

bigbrett, 2023-05-18:
Hi @jandriea,

To add a bit to @embhorn's response, we have a wolfSSL [external C module](https://docs.micropython.org/en/latest/develop/cmodules.html) in the works for micropython that will allow you to effectively "replace" the built-in mbedTLS implementation of the `ussl` module with a wolfSSL implementation. We can't add code to micropython directly because our GPLv2 license isn't compatible with MIT, but an external module is the next best thing. Currently, the prototype module supports just the Unix port, but an STM32 port supporting a few common micropython boards is in active development. We will host the module on our GitHub and notify users with a blog post once it is ready to be used!

Regarding wolfSSH: as far as I know, micropython itself does not have any support for SSH at all. We likely wouldn't contribute an SSH implementation upstream unless there is significant customer demand, but it isn't out of the question! I imagine once micropython has built-in SSH support, configuring it to use wolfSSH would be much simpler, and something we would likely do. 

Hope this helps!

bigbrett, 2023-05-18:
Also, if all you are looking for is remote access for your device, have you checked out [webREPL](https://github.com/micropython/webrepl)?

jandriea, 2023-05-19:
Hi @embhorn and @bigbrett, thank you for your responses

> Thanks for contacting wolfSSL Support. We are currently working on a wolfSSL implementation of MicroPython. Would you be interested in reviewing when it is ready?

I would love to, please let me know when it's ready.

> wolfSSH is tightly integrated with the wolfCrypt library. It might be possible to use with mbedTLS, but not something we would be interested in supporting. That said, one of our engineers started doing some preliminary work on mapping MBEDTLS calls to wolfSSL calls, you can review those alpha changes here: [wolfSSL/wolfssl@master...dgarske:mbedtls](https://github.com/wolfSSL/wolfssl/compare/master...dgarske:mbedtls)

Actually, using mbedTLS is not a must. I use it because it is already supported in the MicroPython project. I am open to switching to any other TLS solution if it works with MicroPython.

> To add a bit to @embhorn's response, we have a wolfSSL [external C module](https://docs.micropython.org/en/latest/develop/cmodules.html) in the works for micropython that will allow you to [...]

This is great news! I hope the development process will run smoothly

> Regarding wolfSSH: as far as I know, micropython itself does not have any support for SSH at all. We likely wouldn't contribute an SSH implementation upstream unless there is significant customer [...]

The MicroPython does not support SSH, however, there is a fork that successfully implemented the SSH module in ESP32 board. Sadly, I am using a different MicroPython port, so there are still a lot of things to do to make it works with the STM32 board.

> Also, if all you are looking for is remote access for your device, have you checked out [webREPL](https://github.com/micropython/webrepl)?

As to this question, the project that I'm working on needs an ssh client that supports SCP protocol to transfer and receive data, and also to send some commands to the ssh server. 

Again, thank you for your help. I will look forward to the external module that you are working on. Hopefully, it will come in the near future.

closed 2023-05-27 by jandriea