wolfSSL/wolfssh #562: diffie-hellman-group-exchange-sha256 broken since v1.4.12-stable

falemagn, 2023-08-02:
Hi,

Commit 3e1a6ff3 introduced a bug that makes KEX with diffie-hellman-group-exchange-sha256 not work any longer.

More precisely, the `primeGroup` variable used as the first parameter to `CreateMpint` at line [8343 in internal.c](https://github.com/wolfSSL/wolfssh/blob/3e1a6ff34255e045f1811637cc060d1f3d23276b/src/internal.c#L8343) is always NULL, resulting in `CreateMpint` returning with an error.

`primeGroup` is NULL because the call to `CreateMpint` has been moved to a new function named [`SendKexGetSigningKey` ](https://github.com/wolfSSL/wolfssh/blob/3e1a6ff34255e045f1811637cc060d1f3d23276b/src/internal.c#L8079), but the code that was setting up  `primeGroup` has been left [in the original function](https://github.com/wolfSSL/wolfssh/blob/3e1a6ff34255e045f1811637cc060d1f3d23276b/src/internal.c#L8511).

You can easily verify the issue with the echoserver and OpenSSH's sftp.

In one console issue the following commands:
```
cd examples/echoserver
./echoserver -E -d /
```

In another console, issue the following commands:
```
sftp  -P 22222 -oKexAlgorithms=diffie-hellman-group-exchange-sha256 jill@localhost
```

You will see the KEX starting and then the connection closing abruptly.

In the echoserver's log, you will see this:
```
[...]
[DEBUG] Decoding MSGID_KEXDH_GEX_INIT
[DEBUG] Entering SendKexDhReply()
[ERROR] Internal argument error with CreateMpint
```

JacobBarthelmeh, 2023-08-03:
Thanks @falemagn! Reproduced the reported issue. Am working on a fix.

JacobBarthelmeh, 2023-08-09:
Fix was in PR566 and merged in. Thanks again for the report!

closed 2023-08-09 by JacobBarthelmeh