wolfSSL/wolfssh #564: non blocking mode does not work : infinte lines of "client would read block" in offical test client program

AnounIssues, 2023-08-05:
as the title says non blocking mode does not work 
it prints infinite number of "... client would read block"s 
test both in visual studio using msvc compiler and in Linux using gcc

wolfSSH_connect return WS_FATAL_ERROR one time (which it should not ) and it's ignored by NonBlockSSH_connect which causes infinite loop 

acutely it's to bugs : 1-wolfSSH_connect  return WS_FATAL_ERROR on non blocking mode for an ip+user+pass which successfully connects in blocking mode 
2-NonBlockSSH_connect  in official test client program ignores the error and get stock at infinite loop 

the cause for WS_FATAL_ERROR is a socket error in one of the functions wolfSSH_connect calls , i guess the function is forgotten to be made compatible with non blocking mode

how to reproduce :
client -h 127.0.0.1 -u USER -p 22 -P PASS -> works and connects successfully 
client -h 127.0.0.1 -u USER -p 22 -P PASS -N -> does not work. infinite loop

JacobBarthelmeh, 2023-08-07:
This should fix the client side connect in nonblocking cases (https://github.com/wolfSSL/wolfssh/compare/master...JacobBarthelmeh:blocking?expand=1#diff-3da5da4d7415361a398fcf93e1863d0ae8c74a7075dbe283099104361ff5cd7cR156). Can you confirm that it resolves the issue for you too?

AnounIssues, 2023-08-09:
@JacobBarthelmeh 
unfortunately  it does not fix
i guess you have test with wolfsshd as server (because you made changes to wolfsshd)
ssh is an standard i recommend testing against openssh as wildly used implementation 
i tested against my ubuntu vps with default ssd config and it failed similarly  ( putty and bitvisessh client  do connect with no problem)

AnounIssues, 2023-08-12:
@JacobBarthelmeh hi again , sorry for late reply
in
"$ ./examples/client/client -u <user> -p 22 -h 127.0.0.1 -t"
you are not using non blocking mode , please add -N option
without -N it connects successfully for me too 

"Is the socket error from the peer closing down when the client is attempting to connect?"
unfortunately i can't test as i'm away now, if the question is about whether the servers responses properly , i think when wolfssh client connect successfully on blocking mode  and fails on non blocking and other popular clients connect successfully there is a bug in non blocking mode ,not the server

JacobBarthelmeh, 2023-08-14:
@AnounIssues you're right, I missed the -N nonblocking flag in that previous test for reproducing. I am reproducing an issue now and working on a fix.

AnounIssues, 2023-08-19:
@JacobBarthelmeh good, thanks for the schatual

JacobBarthelmeh, 2023-08-25:
@AnounIssues got back to this report and think I have a proper fix now (https://github.com/wolfSSL/wolfssh/pull/565/commits/bd1506db932b3a60c9e70419da9357b2297d237c). Thanks for your patience on this!

Could you tell us some about your projects use case with wolfSSH? It's always interesting to hear how the library gets used.

closed 2023-08-28 by ejohnstown