wolfSSL/wolfssh #576: make fail with sftp

madz0, 2023-09-11:
Hi! when trying to build with sftp enabled:
`./configure --enable-debug --enable-certs --enable-static --disable-examples --enable-sftp`
and then run `make` I get:
```
src/internal.c: In function ‘AddAssign64’:
src/internal.c:13543:23: error: ‘WOLFSSL_MAX_32BIT’ undeclared (first use in this function); did you mean ‘WOLFSSL_MAX_16BIT’?
     if (addend1[0] > (WOLFSSL_MAX_32BIT - addend2)) {
                       ^~~~~~~~~~~~~~~~~
                       WOLFSSL_MAX_16BIT
```
Looks like `WOLFSSL_MAX_32BIT` is not defined anywhere, but `WOLFSSL_MAX_16BIT` is

anhu, 2023-09-11:
Hi, 

That is very odd.  They are both defined here: 

https://github.com/wolfSSL/wolfssl/blob/63477bcfdb8fee23a60bf3f7fd92ccb2911d6715/wolfssl/wolfcrypt/types.h#L298
https://github.com/wolfSSL/wolfssl/blob/63477bcfdb8fee23a60bf3f7fd92ccb2911d6715/wolfssl/wolfcrypt/types.h#L299

This is for our master branch of the wolfSSL repo.  What version of wolfSSL are you building against? 

Warm regards, Anthony

madz0, 2023-09-12:
Hi @anhu you're right! thanks for the reply. I got wolfssh master but was compiling against an old wolfssl

closed 2023-09-12 by madz0