wolfSSL/wolfssh #594: Double Free or Corruption on 1.4.13 and master branch

busmaker, 2023-09-28:
Env
---
- Ubuntu 22.04
- Gcc 13.1.0
- Wolfssl 5.6.3
- Wolfssh 1.4.13 and master branch

Building
---

Wolfssl built using cmake
```sh
cmake -S /projects/cpp/gamers/3rdparty/wolfssl-5.6.3/ -B . -DCMAKE_INSTALL_PREFIX=/sources/wolfssl/wolfssh-1.4.13-stable/ -DWOLFSSL_KEYGEN=ON -DBUILD_SHARED_LIBS=OFF

make -j16
make install
```

Wolfssh built with
```sh
./autogen.sh

./configure -with-wolfssl=/sources/wolfssl/wolfssh-1.4.13-stable --enable-static=yes --enable-shared=no --enable-sshd --enable-keygen CFLAGS="-g" LIBS=-lm

make check // 3 tests failed for the same reason below.
```
---

After successful build, execute `./examples/echoserver/echoserver` will crash with the error output:
```
double free or corruption (out)
Aborted (core dumped)
```

Here is the call stack:
```
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737350608704) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737350608704) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737350608704, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff7ce7476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff7ccd7f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff7d2e6f6 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7e80b8c "%s\n")
    at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007ffff7d45d7c in malloc_printerr (str=str@entry=0x7ffff7e837b0 "double free or corruption (out)")
    at ./malloc/malloc.c:5664
#7  0x00007ffff7d47ef0 in _int_free (av=0x7ffff7ebec80 <main_arena>, p=0x5555556249c0, have_lock=<optimized out>)
    at ./malloc/malloc.c:4588
#8  0x00007ffff7d4a4d3 in __GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3391
#9  0x00005555555981e2 in wolfSSL_Free ()
#10 0x000055555555f140 in IdentifyKey (in=in@entry=0x555555624520 "0\202\004\243\002\001", inSz=inSz@entry=1191,
    isPrivate=isPrivate@entry=1, heap=0x0) at src/internal.c:957
#11 0x000055555555f2fc in wolfSSH_ProcessBuffer (ctx=ctx@entry=0x555555624480,
    in=in@entry=0x7fffffffdc80 "0\202\004\243\002\001", inSz=inSz@entry=1191, format=format@entry=0, type=type@entry=2)
    at src/internal.c:1152
#12 0x000055555555cbea in wolfSSH_CTX_UsePrivateKey_buffer (ctx=ctx@entry=0x555555624480,
    in=in@entry=0x7fffffffdc80 "0\202\004\243\002\001", inSz=1191, format=format@entry=0) at src/ssh.c:1620
#13 0x0000555555558b48 in echoserver_test (args=args@entry=0x7fffffffe190) at examples/echoserver/echoserver.c:2331
#14 0x0000555555559e14 in wolfSSH_Echoserver (argc=<optimized out>, argv=<optimized out>)
    at examples/echoserver/echoserver.c:2646
#15 0x00007ffff7cced90 in __libc_start_call_main (main=main@entry=0x555555557980 <main>, argc=argc@entry=1,
    argv=argv@entry=0x7fffffffe2f8) at ../sysdeps/nptl/libc_start_call_main.h:58
#16 0x00007ffff7ccee40 in __libc_start_main_impl (main=0x555555557980 <main>, argc=1, argv=0x7fffffffe2f8,
    init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe2e8)
    at ../csu/libc-start.c:392
#17 0x00005555555579b5 in _start ()
```

ejohnstown, 2023-09-28:
I'm looking into this. Thanks for the report.

ejohnstown, 2023-09-28:
Something in the Cmake build of wolfSSL is making both wolfSSL and wolfSSH think that RsaKeys are different sizes. My workaround for this issue for the moment is to use configure to set up wolfSSL. I'll get back to you on Cmake.

ejohnstown, 2023-10-30:
@busmaker, may I ask about your project and who you are working with?

closed 2023-10-31 by ejohnstown

busmaker, 2023-11-10:
> @busmaker, may I ask about your project and who you are working with?

Sorry for the late reply, it's just a personal project, no big deal I found a solution. Thanks.