wolfSSL/wolfssh #659: Runtime configurability of key signature algorithms for KEX

falemagn, 2024-02-22:
The available algorithms can currently be configured only at compile time. It would be useful to have the possibility to compile them in, but enable/disable them at runtime in the context of KEX.

So that:
1. The server_host_key_algorithms namelist in SSH_MSG_KEXINIT includes the algorithm
2. If the algorithm is agreed upon by the client and server, then it's used to sign the key

The specific use case that prompted this feature request is the usage ssh-rsa host keys: we'd like to be able to configure at runtime whether or not they can be signed with ssh-rsa or ssh-sha2-256 (and in the future ssh-sha2-512).

dgarske, 2024-02-22:
See ZD 17568

falemagn, 2024-02-22:
Cannot access that

dgarske, 2024-02-22:
I was just adding that information for the assignment to @ejohnstown. @falemagn that's the ticket you filed in our Zendesk.

falemagn, 2024-02-22:
There must be a misunderstanding. As said, I've got no access to that ticket, because I have personally not filed it. If this conversation must be moved there, I suppose you need to do something to involve me in it.

ejohnstown, 2024-02-22:
You need to set the preprocessor define `WOLFSSH_YES_SSH_RSA_SHA1`. The code for `ssh-rsa` using SHA-1 is still available.

We disabled SHA-1 when OpenSSH announced that they were disabling it. SHA-1 is broken and provides no security. We decided to disable it as well.

falemagn, 2024-02-22:
It's clear that it can be enabled at compile time - that's the premise of [this very issue](https://github.com/wolfSSL/wolfssh/issues/659#issue-2148654026):

> The available algorithms can currently be configured only at compile time.

What I am proposing is to be able to disable/enable **at runtime** the  algorithms that have been compiled in, just like it happens with OpenSSH.

Btw, the [release note](https://www.wolfssl.com/wolfssh-1-4-14-release/) of the version that disabled the SHA1 signature didn't mention it. Was it mentioned anywhere else?

ejohnstown, 2024-03-13:
I'm going to close this issue since the PR adds the requested functions. I believe it achieves the desired goal.

closed 2024-03-13 by ejohnstown