wolfSSL/wolfssh #697: wolfssh_template / wolfssh_echoserver don't build

keredson, 2024-05-21:
to replicate:
```
$ idf.py --version
ESP-IDF v5.2.1
$ idf.py create-project-from-example "wolfssl/wolfssh^1.4.15-stable-preview6:wolfssh_template"
[...]
$ cd wolfssh_template/
$ idf.py build
[...]
/tmp/wolfssh_template/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/port/Espressif/esp32-crypt.h:890:10: error: #error "ESP32_CRYPT_HASH_SHA256 not supported on wolfSSL at this time"
  890 |         #error "ESP32_CRYPT_HASH_SHA256 not supported on wolfSSL at this time"
      |          ^~~~~
ninja: build stopped: subcommand failed.
```

same error for `wolfssh_echoserver`.

keredson, 2024-05-21:
in `./components/wolfssl__wolfssl/include/user_settings.h`, comment out and add the following:
```
//#define WOLFSSL_SHA224
#define NO_WOLFSSL_ESP32_CRYPT_HASH_SHA256
```
and it works.

embhorn, 2024-05-21:
Hi @keredson 

Thanks for this report. Could you tell us a bit about your project using wolfSSH?

I've requested a review of this issue by our engineers.

gojimmypi, 2024-05-21:
Hi @keredson and @embhorn 

I've been able to reproduce this problem. The issue was related to concurrent SHA HW/SW interleave calculations and fixed in post 5.7.0 release in https://github.com/wolfSSL/wolfssl/pull/7505.

As an interim solution:

* rename `wolfssh_template/managed_components` to `wolfssh_template/components`
* edit the `CMakeLists.txt` in `wolfssh_template/components/wolfssl__wolfssl`. Uncomment and set `WOLFSSL_ROOT`:

```
# Optionally set your source to wolfSSL in your project CMakeLists.txt like this:
set(WOLFSSL_ROOT "/mnt/c/workspace/wolfssl-gojimmypi-pr" )
```

Set the  `WOLFSSL_ROOT` to the fully qualified path of a local, fresh clone of [wolfssl](https://github.com/wolfSSL/wolfssl/tree/master).

I will update the published [wolfSSL Managed Component](https://components.espressif.com/components/wolfssl/wolfssl) this week.

keredson, 2024-05-21:
🎉

closed 2024-05-21 by keredson

gojimmypi, 2024-05-22:
Hi @keredson

I see you've already closed this issue. Not sure if perhaps you saw some of my work in progress at the Expressif Registry. 

I've updated the published components as noted in https://github.com/wolfSSL/wolfssh/pull/699

wolfSSH: https://components.espressif.com/components/wolfssl/wolfssh/versions/1.4.17-preview1m5

and

wolfSSL: https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.1-preview2d?language=en

I believe the examples should be working much better now. Please take it for a test drive and let me know how it goes.

The updated (preview) echoserver can be found here:

https://components.espressif.com/components/wolfssl/wolfssh/versions/1.4.17-preview1m5/examples/wolfssh_echoserver

```
idf.py create-project-from-example "wolfssl/wolfssh^1.4.17-preview1m5:wolfssh_echoserver"
```

Thank you.