wolfSSL/wolfssh #737: wolfsftp.c: WS_FATAL_ERROR upon WS_WANT_WRITE

dgoldbr, 2024-09-01:
Hi!

I am wondering - ins't WS_WANT_READ/WRITE a non-fatal error by design? See here:
https://github.com/wolfSSL/wolfssh/blob/ceb461897bcc73390ba4ac3be14f5ca0bb87fda2/src/wolfsftp.c#L411-L419

But what about this:
https://github.com/wolfSSL/wolfssh/blob/ceb461897bcc73390ba4ac3be14f5ca0bb87fda2/src/wolfsftp.c#L1573-L1577

embhorn, 2024-09-02:
Hi @dgoldbr 

I have requested a review of this topic by our engineers.

ejohnstown, 2024-09-12:
WS_FATAL_ERROR and WS_ERROR are the same value.

ejohnstown, 2025-12-05:
The story is I wanted to just use WS_ERROR and quit using WS_FATAL_ERROR, since either way you check the stored error code and decide what to do. I never got back to it. I'm thinking it should be WS_ERROR and WS_STATUS, but either way you should still look at the stored error code. This is on the list of things to revisit.