wolfSSL/wolfssh #772: piping data over wolfsshd stalls/fails

Forza-tng, 2025-02-10:
Hi,

I am having difficulties piping data from a openssh client to a  wolfsshd host.

For example, the following command never finishes.

```
cat test | ssh 192.168.0.1 -p 1234 'cat > xxxxxxxx'
```
ctrl-c aborts the connection. The test file is 48894 bytes, but 8188 bytes only makes it to the `xxxxxxx` file. The same happens using `mbuffer -i file ...` 

It seems to always be `8188` bytes, even if I cat a large ISO or some text files. However when doing `cat /dev/random` or `/dev/urandom`, the command finishes but no remote file is created at all.

I built wolfssh from a git checkout today using `./configure --prefix=/opt/wolfssh --enable-all --with-wolfssl=/opt/wolfssl`

I use the following `sshd_config`
```
Port 1234
PermitRootLogin yes
PasswordAuthentication yes
LoginGraceTime 30
HostKey /opt/wolfssh/etc/host_key_ed25519.pem
HostKey /opt/wolfssh/etc/host_key_ecdsa.pem
HostKey /opt/wolfssh/etc/host_key_rsa.pem
```
I start wolfsshd using ` /opt/wolfssh/bin/wolfsshd -d -E /tmp/wolfsshd.log  -f /opt/wolfssh/etc/sshd_config`

Logfile contains
```
...
[PID 67002]: [SSHD] Incoming TCP data found
[PID 81519]: [SSHD] Checking user name root
[PID 81519]: [SSHD] Unix check user
[PID 81519]: [SSHD] User ok.
[PID 81519]: [SSHD] Attempting to raise permissions level
[PID 81519]: [SSHD] Public key ok.
[PID 81519]: [SSHD] Lowering permissions level
[PID 81519]: [SSHD] Checking user name root
[PID 81519]: [SSHD] Unix check user
[PID 81519]: [SSHD] User ok.
[PID 81519]: [SSHD] Attempting to raise permissions level
[PID 81519]: [SSHD] Public key ok.
[PID 81519]: [SSHD] Lowering permissions level
[PID 81519]: [SSHD] grace time = 30 timeout = 0
[PID 81519]: [SSHD] Entering exec session [cat > /xxxxxxxx]
[PID 81519]: [SSHD] Attempting to raise permissions level
[PID 81519]: [SSHD] Attempting to close down connection
[PID 81519]: [SSHD] Return from closing connection = 0
```

The source test file was created using the following, so I can determine how much was processed.
```
for i in {1..10000} ; do echo $i ; done >> test
```

An strace of wolfsshd reveals it receives and sends 1859 rows and then stops with `<unfinished ...>` reading from `/dev/pts/9<char 136:9`

```
83782<cat> 01:15:52.827126 read(0</dev/pts/9<char 136:9>>, "1857\n", 262144) = 5
83782<cat> 01:15:52.827148 write(1</xxxxxxxx>, "1857\n", 5) = 5
83782<cat> 01:15:52.827172 read(0</dev/pts/9<char 136:9>>, "1858\n", 262144) = 5
83782<cat> 01:15:52.827194 write(1</xxxxxxxx>, "1858\n", 5) = 5
83782<cat> 01:15:52.827218 read(0</dev/pts/9<char 136:9>>, "1859\n", 262144) = 5
83782<cat> 01:15:52.827240 write(1</xxxxxxxx>, "1859\n", 5) = 5
83782<cat> 01:15:52.827264 read(0</dev/pts/9<char 136:9>>,  <unfinished ...>
83405<wolfsshd> 01:15:53.642975 <... pselect6 resumed>) = 0 (Timeout)
83405<wolfsshd> 01:15:53.643025 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
83405<wolfsshd> 01:15:54.643890 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
83405<wolfsshd> 01:15:55.645052 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
```
[... 85 lines cut ...]
86758<wolfsshd> 01:36:41.727556 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
86758<wolfsshd> 01:36:42.728104 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
86758<wolfsshd> 01:36:43.729116 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
86758<wolfsshd> 01:36:44.730267 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
87232<cat> 01:36:33.640770 write(1</xxxxxxxx>, "278\n", 4) = 4
87232<cat> 01:36:33.640797 read(0</dev/pts/9<char 136:9>>, "279\n", 262144) = 4
87232<cat> 01:36:33.640823 write(1</xxxxxxxx>, "279\n", 4) = 4
87232<cat> 01:36:33.640850 read(0</dev/pts/9<char 136:9>>, "280\n", 262144) = 4
87232<cat> 01:36:33.640876 write(1</xxxxxxxx>, "280\n", 4) = 4
87232<cat> 01:36:33.640903 read(0</dev/pts/9<char 136:9>>, "281\n", 262144) = 4
87232<cat> 01:36:33.640928 write(1</xxxxxxxx>, "281\n", 4) = 4
87232<cat> 01:36:33.640955 read(0</dev/pts/9<char 136:9>>, "282\n", 262144) = 4
87232<cat> 01:36:33.640979 write(1</xxxxxxxx>, "282\n", 4) = 4
87232<cat> 01:36:33.641005 read(0</dev/pts/9<char 136:9>>, "283\n", 262144) = 4
87232<cat> 01:36:33.641028 write(1</xxxxxxxx>, "283\n", 4) = 4
87232<cat> 01:36:33.641052 read(0</dev/pts/9<char 136:9>>,  <unfinished ...>
86758<wolfsshd> 01:36:33.719328 <... pselect6 resumed>) = 0 (Timeout)
86758<wolfsshd> 01:36:33.719410 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
86758<wolfsshd> 01:36:34.719832 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
86758<wolfsshd> 01:36:35.720935 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
86758<wolfsshd> 01:36:36.722026 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
86758<wolfsshd> 01:36:37.723281 pselect6(5, [4<TCP:[0.0.0.0:1234]>], [], [], {tv_sec=1, tv_nsec=0}, NULL) = 0 (Timeout)
```

I'm running in Gentoo Linux, x86_64, gcc-14.2.1. I did try this in September last year, with the same issue.

JacobBarthelmeh, 2025-02-11:
Thanks for the  detailed report. As a note I was able to reproduce the reported behavior.

JacobBarthelmeh, 2025-02-17:
@Forza-tng a fix for this report is under review here (https://github.com/wolfSSL/wolfssh/pull/776).

Forza-tng, 2025-02-22:
Thanks. With the latest commit, it now works :)

Forza-tng, 2025-02-26:
Compiling wolfssl with aesni and asm made a big difference.

For aes256-gcm both wolfsshd and opensshd reach approx 1.3GiB/s.
For aes256-ctr wolfsshd reach 400MiB/s while opensshd stays at 1.3GiB/s

The higher numbers compared to before is because the system was under load from other things. Today I was testing with an unloaded system.