wolfSSL/wolfssh #780: Inconsistent Directory Naming: wolfssh/ide vs wolfssl/IDE

clutch2sft, 2025-02-19:
Description:
I’ve observed that the directory structures for wolfssh and wolfssl use different naming conventions for their IDE folders. Specifically, wolfssh places its IDE-related files in wolfssh/ide/ (all lowercase), while wolfssl uses wolfssl/IDE/ (uppercase).

Impact:
This discrepancy could lead to confusion among contributors and might cause issues on case-sensitive file systems. Consistent naming conventions are generally considered a best practice for maintainability and clarity, as noted in various coding style guidelines (e.g., Google's Style Guides and similar resources).

Suggestion:
It might be beneficial to standardize the naming convention across both repositories. Aligning the directory names—either both as ide or both as IDE—would improve consistency and reduce the potential for errors.

embhorn, 2025-02-19:
Hello @clutch2sft 

Thanks for your interest in wolfSSH. Could you tell us a bit about your project?

I agree that consistency would be helpful. I'll run this past the team, as it would be a relatively quick update.

Thanks,
@embhorn - wolfSSL Support

gojimmypi, 2025-02-19:
Although it is indeed unfortunate that the case of `IDE` is inconsistent between the repositories, and the change would be quick, all of the respective links in scripts, code examples, and issue code references would need to be edited as the GitHub URL paths are case sensitive.  See:

working: https://github.com/wolfSSL/wolfssh/tree/master/ide/Espressif

vs

404: https://github.com/wolfSSL/wolfssh/tree/master/IDE/Espressif

clutch2sft, 2025-02-19:
@embhorn Right now I needed an sshclient for some top secret project :-)  At the end of the day it is simply a way for me to connect to network devices in a secure manner.  But rather than trying to roll my own, and all the complexity that comes with that  I found this beautifully coded solution so trying to see if it will work.  It's just a playground dream of mine right now but I'll be happy to keep you updated if it gets out of my head into a real project.

@gojimmypi I figured it wouldn't be easy... Just wanted to call it out to gauge the interest.  It is just a niggle for me in having to remember which way to make my pinky finger go when traversing the directories.

embhorn, 2025-02-19:
@clutch2sft 

Thanks for the project details. I'll go ahead and close this issue.

closed 2025-02-19 by embhorn