wolfSSL/wolfssh #781: wolfssh_echoserver Example Fails to Compile Due to wc_SSH_KDF Implicit Declaration

clutch2sft, 2025-02-19:
When compiling the wolfssh_echoserver example located at wolfssh/ide/Espressif/ESP-IDF/examples/wolfssh_echoserver, the build fails with the following error:

```
~/wolfssh/src/internal.c: In function 'GenerateKey':
~/wolfssh/src/internal.c:2215:15: error: implicit declaration of function 'wc_SSH_KDF'; did you mean 'wc_HKDF'? [-Wimplicit-function-declaration]
 2215 |         ret = wc_SSH_KDF(hashId, keyId, key, keySz,
      |               ^~~~~~~~~~
      |               wc_HKDF
```
t appears that wc_SSH_KDF is conditionally compiled behind a feature flag. The only workaround I found was to add the following definition in components/wolfssl/include/user_settings.h:

```
#define WOLFSSL_WOLFSSH
```
Steps to Reproduce:

Clone the repository.
Navigate to wolfssh/ide/Espressif/ESP-IDF/examples/wolfssh_echoserver.
Attempt to build the example using ESP-IDF.
Observe the compile error in wolfssh/src/internal.c regarding wc_SSH_KDF.
Observed Behavior:
The build fails because wc_SSH_KDF is implicitly declared, suggesting that the corresponding feature is not enabled by default.

Expected Behavior:
The example should compile out-of-the-box without requiring manual modifications. If enabling the SSH functionality via a flag like WOLFSSL_WOLFSSH is intended, the documentation should clearly state this requirement.

Potential Workaround:
Adding the following line in components/wolfssl/include/user_settings.h resolves the compilation error:

```
#define WOLFSSL_WOLFSSH
```
Request for Clarification/Action:

Could the maintainers confirm if this is the intended configuration for building the wolfssh_echoserver example?
If so, would it be possible to update the documentation or the build configuration to automatically define WOLFSSL_WOLFSSH for this example?
Alternatively, if there is another configuration method intended for this setup, please provide guidance on how to correctly enable the feature.
Thank you for your time and consideration. I’m happy to help further if needed!

LinuxJedi, 2025-02-19:
@gojimmypi can you please take a look at this?

clutch2sft, 2025-02-19:
@gojimmypi 
Thank you for such a detailed response.  I'm glad you're on top of these things.  As for now I had given up on managed component (though I am in love with the idea of a managed component) as I wasn't moving my toward my objective fast enough with that.  

The staging thing is interesting I'll keep an eye on that - I'm just learning my way around this esp platform at the same time I'm learning my way around the wolfssl/ssh components.  Happy to circle back and test it as time permits!  With the of WOLFSSL_WOLFSSH I am able to keep my own project going forward, well that and the fix I put in the directory search stuff.

Again truly apricate the thoughtful response and your work.  I'm not married to my fixes making it into production on the repo but I like to share the information incase it is helpful since so much effort is put out  on your parts.

gojimmypi, 2025-02-24:
Hi @clutch2sft - As noted in the related https://github.com/wolfSSL/wolfssh/issues/782, I've updated the Managed Components for wolfSSL to [v5.7.6](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.6) and for wolfSSH to [1.4.18](https://components.espressif.com/components/wolfssl/wolfssh/versions/1.4.18).

The [wolfssh_echoserver example](https://components.espressif.com/components/wolfssl/wolfssh/versions/1.4.18/examples/wolfssh_echoserver?language=en) works for me, tested on ESP-IDF v5.2. I was hoping you could take it for a test drive and confirm before I upgrade the wolfSSH library to the latest release.

closed 2025-12-15 by douzzer