wolfSSL/wolfssh #782: Standalone Project Setup Fails: SRC_DIRS Error When Extracting wolfssh_echoserver Example

clutch2sft, 2025-02-19:
Description:
The wolfssh_echoserver example is presented as a template to simplify the integration of wolfssh capabilities into custom ESP-IDF projects. However, when attempting to extract this example and set it up as a standalone project, the build process fails with the following error:

`CMake Error at ~/esp/v5.4/esp-idf/tools/cmake/component.cmake:285 (message):
  SRC_DIRS entry '~/src/' does not exist.
Call Stack (most recent call first):
 ~/esp/v5.4/esp-idf/tools/cmake/component.cmake:454 (__component_add_sources)
  components/wolfssh/CMakeLists.txt:437 (idf_component_register)
`
Steps to Reproduce:

Extract or copy the wolfssh_echoserver example from wolfssh/ide/Espressif/ESP-IDF/examples/wolfssh_echoserver to a new directory outside of the original repository tree.
Set up a new ESP-IDF project using the extracted example.
Run the build process with ESP-IDF.
Encounter the SRC_DIRS error during the CMake configuration.
Observed Behavior:
The build fails because the CMake configuration for the wolfssh component references a directory path (~/src/) that does not exist in the new project structure.

Expected Behavior:
As a template, the wolfssh_echoserver example should be easily extractable and configurable as a standalone project. The build system should use relative paths that are valid in the context of the new project location.

Potential Considerations:

The error might be related to how the SRC_DIRS entry is defined in the components/wolfssh/CMakeLists.txt.
Adjusting the SRC_DIRS configuration to reference paths relative to the project root could resolve this issue.
Request for Guidance/Action:

Could the maintainers provide clarification on the intended setup for the wolfssh_echoserver example when used as a standalone project?
Is there a recommended modification or configuration change (e.g., adjusting the SRC_DIRS path) that would allow the example to build successfully outside of the main repository tree?
Alternatively, should the example documentation be updated to include instructions for setting up the project independently?
Thank you for your time and assistance. I look forward to any guidance or suggestions you might have as I work through this issue.

LinuxJedi, 2025-02-19:
@gojimmypi can you please take a look at this one too?

clutch2sft, 2025-02-19:
I was able to resolve this so that the code works in both cases - but I am not very adept with git so just giving you my fixes here.  Hopefully it helps.

```
# ---------------------------------------------------------------------------
# Updated IS_WOLFSSH_SOURCE:
# This function expands a leading "~" in the provided directory and checks 
# whether the resulting directory contains the file "wolfssh/ssh.h". 
# (This is the expected layout for the wolfSSH repository where the header 
# lives in a subdirectory named "wolfssh".)
#
# If the file is found, the directory is considered a valid wolfSSH source tree.
# Otherwise, it returns an empty result.
# ---------------------------------------------------------------------------
function(IS_WOLFSSH_SOURCE DIRECTORY_PARAMETER RESULT)
    # Expand a leading tilde, if present.
    if("${DIRECTORY_PARAMETER}" MATCHES "^~")
        string(REGEX REPLACE "^~" "$ENV{HOME}" ABS_DIR "${DIRECTORY_PARAMETER}")
    else()
        set(ABS_DIR "${DIRECTORY_PARAMETER}")
    endif()

    message(STATUS "IS_WOLFSSH_SOURCE: Checking directory: ${ABS_DIR}")

    # Look for the header in the expected location.
    if(EXISTS "${ABS_DIR}/wolfssh/ssh.h")
        message(STATUS "IS_WOLFSSH_SOURCE: Found ssh.h in ${ABS_DIR}/wolfssh/ssh.h")
        set(${RESULT} "${ABS_DIR}" PARENT_SCOPE)
    else()
        message(STATUS "IS_WOLFSSH_SOURCE: ssh.h not found in ${ABS_DIR}/wolfssh/ssh.h")
        set(${RESULT} "" PARENT_SCOPE)
    endif()
endfunction()

# ---------------------------------------------------------------------------
# Updated FIND_WOLFSSH_DIRECTORY:
# This function determines the wolfSSH source root as follows:
#
# 1. If a cached WOLFSSH_ROOT variable is defined (for example, "~/wolfssh"),
#    it expands any leading "~" into the full home directory path and calls
#    IS_WOLFSSH_SOURCE to check if that directory contains "wolfssh/ssh.h".
#    If it does, that directory is used as the wolfSSH source tree.
#
# 2. If no valid WOLFSSH_ROOT is provided, the function searches upward
#    from CMAKE_CURRENT_LIST_DIR (the directory containing the current CMakeLists.txt)
#    until it finds a directory that passes IS_WOLFSSH_SOURCE.
#
# The function returns the found directory (or an empty string if none is found).
# ---------------------------------------------------------------------------
function(FIND_WOLFSSH_DIRECTORY OUTPUT_FOUND_WOLFSSH_DIRECTORY)
    # If a cached WOLFSSH_ROOT is provided and not empty, use it.
    if(DEFINED WOLFSSH_ROOT)
        if(NOT WOLFSSH_ROOT STREQUAL "")
            string(REGEX REPLACE "^~" "$ENV{HOME}" EXPANDED_ROOT "${WOLFSSH_ROOT}")
            set(WOLFSSH_ROOT "${EXPANDED_ROOT}")
            message(STATUS "FIND_WOLFSSH_DIRECTORY: Using cached WOLFSSH_ROOT (expanded): ${WOLFSSH_ROOT}")
            IS_WOLFSSH_SOURCE("${WOLFSSH_ROOT}" FOUND_WOLFSSH)
            if(FOUND_WOLFSSH)
                message(STATUS "FIND_WOLFSSH_DIRECTORY: Cached value is valid: ${FOUND_WOLFSSH}")
                set(${OUTPUT_FOUND_WOLFSSH_DIRECTORY} "${FOUND_WOLFSSH}" PARENT_SCOPE)
                return()
            else()
                message(STATUS "FIND_WOLFSSH_DIRECTORY: Cached WOLFSSH_ROOT (${WOLFSSH_ROOT}) is not a valid wolfssh source tree.")
            endif()
        endif()
    endif()

    message(STATUS "FIND_WOLFSSH_DIRECTORY: WOLFSSH_ROOT not set or invalid; searching upward from ${CMAKE_CURRENT_LIST_DIR}")
    set(CURRENT_SEARCH_DIR "${CMAKE_CURRENT_LIST_DIR}")
    while(NOT CURRENT_SEARCH_DIR STREQUAL "/" AND NOT CURRENT_SEARCH_DIR STREQUAL "")
        message(STATUS "FIND_WOLFSSH_DIRECTORY: Checking ${CURRENT_SEARCH_DIR}")
        IS_WOLFSSH_SOURCE("${CURRENT_SEARCH_DIR}" FOUND_WOLFSSH)
        if(FOUND_WOLFSSH)
            message(STATUS "FIND_WOLFSSH_DIRECTORY: Found wolfssh source in ${CURRENT_SEARCH_DIR}: ${FOUND_WOLFSSH}")
            set(${OUTPUT_FOUND_WOLFSSH_DIRECTORY} "${FOUND_WOLFSSH}" PARENT_SCOPE)
            return()
        endif()
        get_filename_component(CURRENT_SEARCH_DIR "${CURRENT_SEARCH_DIR}" DIRECTORY)
    endwhile()

    set(${OUTPUT_FOUND_WOLFSSH_DIRECTORY} "" PARENT_SCOPE)
endfunction()
```

gojimmypi, 2025-02-19:
@clutch2sft thanks you so much for the suggestion! I'll need to take a closer look at the proposed cmake changes here, but in the meantime please take a look at my update in https://github.com/wolfSSL/wolfssh/pull/773. In particular, see my updates CMakeFiles.txt in the [examples/wolfssh_echoserver](https://github.com/gojimmypi/wolfssh/tree/pr-espressif-examples-770-update/ide/Espressif/ESP-IDF/examples/wolfssh_echoserver).

See also my [#781 response](https://github.com/wolfSSL/wolfssh/issues/781#issuecomment-2669132816).

gojimmypi, 2025-02-24:
Hi @clutch2sft - I've updated the Managed Components for wolfSSL to [v5.7.6](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.6) and for wolfSSH to [1.4.18](https://components.espressif.com/components/wolfssl/wolfssh/versions/1.4.18).

The [wolfssh_echoserver example](https://components.espressif.com/components/wolfssl/wolfssh/versions/1.4.18/examples/wolfssh_echoserver?language=en) works for me, tested on ESP-IDF v5.2. I was hoping you could take it for a test drive and confirm before I upgrade the wolfSSH library to the latest release.

closed 2025-12-18 by embhorn